RaspAP/raspap-webgui #642: Multiple access points

rugk, 2020-07-27:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [ ] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**

Could you possibly add support for multiple access points? If that is possible…
(I know on some OpenWRT devices, it is and you can add many access points there.)

mwelsch, 2020-08-28:
Second this, I'm planning to get a WiFi USB dongle so I can run a 2.4GHz and 5GHz hotspot simultaneously. Any chance this gets implemented?

billz, 2020-08-31:
PRs are welcome :)

rugk, 2020-10-30:
shh #badbot :no_bell::robot::no_bell:

billz, 2020-10-30:
protip: instead of bumping your own feature request, contribute something of value to help push it forward.

rugk, 2020-10-30:
Well, or disable this annoying bot that closes reasonable feature requests. Closing an issue is always declining it. And at least two people seem to be interested in it…

closed 2020-10-30 by billz