RaspAP/raspap-webgui #644: IPv6 support

rugk, 2020-07-28:
https://github.com/billz/raspap-webgui/issues/359 tackled it, but was closed without providing any explanation.

And from what i see, it really all only supports IPv4.

IMHO it would be [really](https://ipv4.rip/) [good](https://ipv6.watch/) [if](https://ipv6wallofshame.com/) [support IPv6](https://ipv4flagday.net/).
(IPv6 flag day is February 1st, 2030; #ipv4flagday)

I guess, this is not really a problem for local networks, of course. However, having support for it, would possibly still be nice.

billz, 2020-07-28:
Good suggestion. What can you contribute?

rugk, 2020-09-27:
shh #badbot :no_bell::robot::no_bell:
bot :roll_eyes: 

This is a feature request…

cmol, 2020-09-30:
So while I haven't tested this, I assume that ipv6 works in bridge mode. If not, that would definitely be the low hanging fruit, along with taking an ipv6 address to the unit itself.

The much harder part is routed mode. The AP needs to know what subnet to deliver addresses in. Typically this is done via DHCPv6-PD, but I'm not sure how many consumer routers that hands out prefixes for internal use.

It's definitely possible, but I'd perhaps look at bridged ipv6 only to begin with as well as obtaining an ipv6 address to access the AP on, and then detecting if a prefix has been delivered via PD.

closed 2020-12-06 by stale[bot]

ajmas, 2021-05-12:
Would it be worth adding in the UI the ability to have different settings for IPv6, such that the bridging rules are available for IPv4 and IPv6 separately? 

For a situation where the servers hands out addresses, there is a choice between DHCPv6 and 'router advertisements' (radvd). I believe the latter is what the IPv6 team prefers, whereas DHCPv6 is what corporations prefer?

One article on using a Raspberry Pi as IPv6 router for a network: 

https://superuser.com/questions/775029/using-raspberry-pi-as-ipv6-router-for-network

TheCataliasTNT2k, 2022-05-05:
Still no ipv6 support? Very sad to hear this.... Then I can not use RaspAP at the moment, and I have to use my own solution until it is implemented. Come on, it is not that hard.... I added it to my scripts in like one day, and never used ipv6 manually before.

billz, 2022-05-06:
@TheCataliasTNT2k now, integrate this into the installer, validate it on each of the supported OS's and do a full regression test. it's not that hard. waiting for your PR 😉

TheCataliasTNT2k, 2022-05-07:
@billz Yes, it is not that hard. I totally agree. And I know that this was irony.
But I never really used PHP and I do not have time for that right now. So yes, technically, it is not that hard.
You just need to add the stuff I did to your setup, and you are done. You do not need to use radvd, you can use DHCPv6, even if it will be deprecated at some point. If you want to, you can test, if the Pi got a complete subnet assigned (as it should per RFC), or you use my method with NAT (even if this should not be done normally, but you can not guarantee, that you get a prefix delegated). Then copy your iptables rules, and you are done. You need to add the needed UI elements and that should be it.
It really that easy.
The only thing, I did not find a good solution right now, is the client list. Without DHCP, that is the only thing, that is more difficult.