RaspAP/raspap-webgui #646: WIFI Hotspot doesn´t come up

boris1000, 2020-08-02:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [ ] I searched existing issues before opening this one
* [ ] I checked the FAQ before creating this issue
* [ ] I have read and understand the issue reporting guidelines 

**Describe the bug**
Just installed RASPAP as given in the official description.
Unfortunately the WIFI AP isn´t coming up.
No error message during installation, ifconfig shows the WLAN0 as 10.3.141.1

**Your environment**
Pi4
Buster
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install or Manual setup? Quick
* Using default configuration? Y
* Simultaneous AP and managed mode? N
* Onboard wireless chipset or external adapter? Onboard
* Other software or services running with RaspAP? Openhab

**Steps to reproduce**
Setup was running with no issues. Now ifconfig shows the WLAN0 as 10.3.141.1
Was rerunning the setup - but wasn´t solving the issue

**Expected behavior**
Expected AccessPoint raspi config coming up to be able to connect to it

**Actual behavior**
There is no AP coming up. On the PC I saw the raspi webgui for a second but disappeared immediately again

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

billz, 2020-08-03:
Sounds like openHAB is creating a conflict. Do a clean install, ie. without openHAB, and report your results. 
Protip: rather than trying to run both applications on the same device, use two Pis.

boris1000, 2020-08-03:
Any idea what could be the problem and what could stopp  the AP coming up?
e.g. I saw that you are replacing dhcpcd.conf, it seems not to disturb Openhab, but are there other things I could maybe check?
It was my plan to have both on one Pi.

boris1000, 2020-08-03:
and Yes I tried it already with raspian light and it was working without problems, same Pi4, also Buster, same RaspAP settings

boris1000, 2020-08-03:
Now I tested with an older version (also Buster) of Openhab and here RaspAP is working. I assume it has something to do with the newest 1.6-alpha version of openhab. So I plan to do a fresh installation using the older Buster base (not Raspberry PiOS based).

boris1000, 2020-08-04:
Again I tried to set it up with Raspberry PiOS and now I got it running, don´t ask me why I did all the same no special configuration the only difference is that I took the 64bit based PiOS.
I fhtis really makes the difference? Doesn´t matter it´s running and I close this issue

closed 2020-08-04 by boris1000