RaspAP/raspap-webgui #649: Getting crazy, AP appears but I cannot connect

boris1000, 2020-08-06:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [ ] I searched existing issues before opening this one
* [ ] I checked the FAQ before creating this issue
* [ ] I have read and understand the issue reporting guidelines 

**Describe the bug**
since days I am trying to get raspap up and running. I am using the bridged mode. I installed already many different Raspian versions on my Pi4 and I got it running on 2.4n on the PiOs64bit but this I don´t want to use cause not yet really stable.
Currently I am on PiOS 32bit and I can get the AP up and running and being able to connect when choosing the 5Ghz.
When chossing 2.4n the AP sometimes appears but I just cannot connect.
Changed already everything for testing but nothing helps!
What can that be?
Is Pi4 2.4n (tested also g) not stable or what is this?

Thanks for your support  

**Your environment**
* Raspberry Pi hardware (examples: Pi 3 Model B+, Pi Zero W)  Pi4
* Raspbian version (examples: Buster Lite, Buster Desktop) PiOS 32bit
* Followed the project prerequisites? (Y/N) y
* Checked the project FAQ? (Y/N) y
* RaspAP Quick Install or Manual setup? Q
* Using default configuration? (Y/N) y
* Simultaneous AP and managed mode? (Y/N) n
* Onboard wireless chipset or external adapter? o
* Other software or services running with RaspAP? openhab

**Steps to reproduce**
Tell us how to reproduce this issue. Provide as much detailed information as possible.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Actual behavior**
Tell us what you observed instead.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

billz, 2020-08-06:
We've established in your previous issue that RaspAP runs perfectly without openHab. Unfortunately, I do not use this software, have never installed it and cannot assist with troubleshooting the conflicts you appear to be having.

You don't mention how openHab is installed. If using Docker, try searching this repo for issues related to it.

boris1000, 2020-08-06:
I see, but I am getting the impression that it is not openhab, it is more that raspap is somehow not correctly following config changes saved and applied. Can it be that it is somhow important in what sequence changes to frequency and bridged or not  are being applied?
This what at least is my impression cause I got it running when changing bridged mode and after that changed frequency and then back to bridged mode ....

billz, 2020-08-06:
Re: accessing the web UI in bridged mode, refer to [this wiki](https://github.com/billz/raspap-webgui/wiki/Bridged-AP-mode). From what I gather, you have no issues connecting to the AP. So I fail to see the problem here.

I would however suggest that you choose a wireless mode, start the AP and enable bridged mode, then reverse this process before switching channels/modes again.

boris1000, 2020-08-06:
and just to confirm, I freshly installed Raspian, installed and configured Openhab, then I installed RaspAP all standard config.
After that I switched to 5G, next is configuring SSID, PW, all works fine.
Still in 5G mode switch bridged mode, still is OK, and now switch to 2.4n and all is OK.

No other way is working, believe me this was now 3 days of sh..., lucky me that I found a way to get it working.

Now I will adjust br0 IP in dhcpcd.conf and hope this won´t disturb :-)

boris1000, 2020-08-06:
So I changed br0 IP, PI is now using the new IP, but I cannot call the WebGui using this new IP, what am I doing wrong?
Is dhcpcd.conf not the right place?

boris1000, 2020-08-06:
seems that the change in dhcpcd.conf was changing the IP from PI but not for RaspAP.
What I did now is using "always assign same IP" in the router and deleted the entries in dhcpcd.conf again.
Now it works, I am connected on 2.4n, can access the RaspAP webgui, and openhab is running in parallel.

OK now my Boss is calling me - I should start with barbecue .....

closed 2020-08-08 by billz