RaspAP/raspap-webgui #650: Feature Request - Additional LAN adapters.

kanaida2, 2020-08-07:
When I configure this as a non bridged mode and it behaves like a router, we don't have a second LAN port on the pi. Let's assume you purchased one.

Now I want the pi to assign an IP and route traffic to both wlan0 and eth1. If I'm not mistaken, all I need to do is add a bridge between wlan0 and eth1. eth0 being the WAN port. The same as a regular router with WiFi and LAN.

My temporary workaround for bridged mode is enabling the hotspot in bridged mode and adding eth1 to the bridge created by raspAP. However this gets my actual router doing the IP assignment. I haven't tested it yet because I don't have a cable.

My workaround for non bridge mode is create a bridge in a boot script with wlan0 and eth1 (along with any other LAN ports). I've yet to test this either because of the cable I don't have until tomorrow.

kanaida2, 2020-08-07:
Perhaps what is needed is a simple page called ports where we can simply say this is WAN, these other adapters are LAN and enabled/disabled ports.

darnokb, 2020-09-08:
This feature would be very usefull. I've ordered NanoPi R1 with two ethernet ports, it would be awesome if I could configure each of them separetely, i.e. one for LAN and one for WAN. If needed I can contribute with tests.

RobertDuncan42, 2020-09-20:
I am looking for similar functionality. 

I want to use my device, for instance, in a camper with a long range USB WiFi device with directional antenna to pick up the weak campground WiFi and rebroadcast that inside the camper as well as deliver ethernet in the camper.

So the ability to select one wlan as the Hotspot AP, a second wlan port as the client up-link to a provider and be able to bridge the AP net to the eth0 port. I would like to be able to configure DHCP on the eth0 for some applications, and not for others.

I don't think a new page is required. Maybe some configuration buttons added to the Networking page per-device tabs.  Allow one port to be set as the up-link, other ports as either Not Used or Access Port.  There is already DHCP controls on the device tabs, so a button to turn on/off DHCP per port.

kanaida2, 2020-09-20:
It turns out it already supported this

closed 2020-11-27 by stale[bot]

FeBe95, 2021-08-21:
Did anyone found a working solution? (for non-bridged mode)
How do I combine wlan0 and eth1 into one bridge in order to use WAN from eth0 and share it with both wlan0 and eth1 (usb adapter)?

kanaida2, 2021-08-21:
I found an even easier way to make a hotspot. Just use Ubuntu server. Then click hotspot in network settings. I think under LAN settings you can select 'shared' or 'shared internet's per device to add them to your internet receiving side along with the wifi.

FeBe95, 2021-08-21:
Thanks for the reply, it's a shame that raspap doesn't support this type of interface configuration... I will be using hostapd with a manual configuration for now, which is obviously lacking the whole Web UI dashboard.