RaspAP/raspap-webgui #651: Feature Request

Alpha-1729, 2020-08-09:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [ ] This is a bug report
* [x] I searched existing issues before opening this one
* [ x] I checked the FAQ before creating this issue
* [ x] I have read and understand the issue reporting guidelines 

**Your environment**
* Raspberry Pi 4B 4GB RAM.  
* Raspbian version :Buster Lite
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install
* Using default configuration? Y
* Simultaneous AP and managed mode? (Y/N)
* Onboard wireless chipset and  external adapter? 
* Other software or services running with RaspAP? No

I have external adapter with high gain antenna. ->wlan1
onboard wireless chipset->wlan0

I need to use wlan1 for wifi client and
wlan0 for  AP.

Is it posssible.
How to set up like this.

billz, 2020-08-09:
https://github.com/billz/raspap-webgui/wiki/FAQs#can-i-use-wlan0-and-wlan1-rather-than-eth0-for-my-ap

closed 2020-08-09 by billz

Alpha-1729, 2020-08-10:
> 
> 
> https://github.com/billz/raspap-webgui/wiki/FAQs#can-i-use-wlan0-and-wlan1-rather-than-eth0-for-my-ap

I need to inverse of that case.
wlan1 as wifi client and wlan0 as AP.