RaspAP/raspap-webgui #652: Feature Request - Wiki explanation for non-technical people

C-Fu, 2020-08-15:
Good day,

I'm trying to figure out how to create a (sometimes) offline router of some sorts with a Pi Zero W + extra usb wifi. 
wlan0: Pi Zero W (connected to my home wifi)
wlan1: Tenda W311U+ / Ralink Technology, Corp. RT2870/RT3070 ("offline" AP for wifi clients)

I would like to have a small router that clients can connect to whenever I'm in the house (with internet connection) or outside (without an internet connection). I read the Wiki but a lot of things just doesn't make sense or seems like it doesn't apply to me. For instance, in the "RPi Zero W AP STA mode" Wiki section nowhere does it state that it's about Internet > wlan0 > wlan1 > clients, so I'm not entirely sure this is what I want to do. But I try anyway, and a new interface, uap0 appeared, but nowhere does it say as such in the wiki. And what about DHCP? What settings should I use? Because when I try to connect with my phone, it stucks at "getting IP address" all the time.

To be clear, I'm pretty sure the wiki is pretty good, especially when compared to a lot of other small projects out there. Just that for a non-techie like me, it's still confusing :D

Also another (small) request: when you click on Start/Restart hotspot, can you make it so it saves the current config first? This might be a deliberate decision, but I think that a start/restart button would imply to save current setting before starting.

Cheers for the awesome work!

billz, 2020-08-17:
AP-STA is only used when you have a single wireless adapter (usually the onboard chipset) and want to configure it as both a wireless client _and_ an AP. Since you have a second adapter on wlan1, [this FAQ](https://github.com/billz/raspap-webgui/wiki/FAQs#can-i-use-wlan0-and-wlan1-rather-than-eth0-for-my-ap) describes a better approach for your setup.

> a new interface, uap0 appeared, but nowhere does it say as such in the wiki

It seems the RPi zero W title got you on the wrong track. This wiki assumes some knowledge of AP-STA mode and virtual interfaces, so I'll add some verbiage to clarify what this is and how it works. 

Re: DHCP, this project's [default configuration](https://github.com/billz/raspap-webgui/wiki/Reporting-issues#default-settings) is designed to get your AP up and running "out of the box". If you're having troubles connecting clients, I'd suggest you check the above FAQ, start with the default settings and enable logging of DHCP requests. Clients may fail to connect for a variety of reasons; the logs will offer clues. The most common problems have been discussed on the [RPi networking forums](https://www.raspberrypi.org/forums/viewforum.php?f=36&sid=8c0ff310f201ce89908a990f560d7d68).

Thereafter, if you want to change IP ranges, DNS servers, configure static leases, etc. all these settings are available to manage.

billz, 2020-08-17:
Done. Hopefully [this is more helpful](https://github.com/billz/raspap-webgui/wiki/RPi-Zero-W-AP-STA-mode).