RaspAP/raspap-webgui #654: Remote Code Execution in Web Interface

lb0x, 2020-08-19:
* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
You don't have guidelines for reporting a security vulnerability, so I don't want to describe in it in too much detail here, as this is public and it shouldn't be disclosed publicly yet, but there is an remote code execution vulnerability in the web interface of RaspAP.

**Your environment**
* Raspberry Pi 4B 4GB
* Raspbian Buster Desktop
* Followed the project prerequisites? (Y/N) Y
* Checked the project FAQ? (Y/N) Y
* RaspAP Quick Install or Manual setup? Quick Install
* Using default configuration? (Y/N) Kinda...
* Simultaneous AP and managed mode? (Y/N) N
* Onboard wireless chipset or external adapter? onboard
* Other software or services running with RaspAP? none.

**Steps to reproduce**
Tell us how to reproduce this issue. Provide as much detailed information as possible.

This is a fairly severe security vulnerability and would rather not disclose it in a public location at this time.

Developers can contact me for more information.

Thank you.

billz, 2020-08-19:
Buster Desktop is not a supported distro.

lb0x, 2020-08-19:
Do you want me to re create it on a supported distro?

You have a fairly glaring RCE that clearly isn't distro dependant.

billz, 2020-08-19:
Emails of project leaders are in the docblock of index.html

lb0x, 2020-08-19:
Email sent. Thanks Bill.

closed 2020-08-19 by lb0x