RaspAP/raspap-webgui #656: Wifi client scan empty in AP-STA mode on RPi 3B

mdeabreu, 2020-08-24:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
Following [these instructions](https://github.com/billz/raspap-webgui/wiki/RPi-Zero-W-AP-STA-mode) to set up an AP-STA mode on a RPi3B, I am able to successfully get the pi connected to my home wifi and produce an access point which connects and provides internet. However clicking the rescan button on the "Wifi Client" page will not show any near by networks (works prior to enabling "Wifi client AP mode"). [Scanning for wifi networks](https://github.com/billz/raspap-webgui/wiki/FAQs#scanning) through ssh does work as expected even though the webgui does not display any results.

**Your environment**
* Raspberry Pi hardware: Raspberry Pi 3B
* Raspbian version: Buster Lite
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install or Manual setup? Quick install
* Using default configuration? Y
* Simultaneous AP and managed mode? Y
* Onboard wireless chipset or external adapter? Onboard only
* Other software or services running with RaspAP? N

**Steps to reproduce**
Prepare SD with latest release of RaspberryPiOS Lite.
Touch `/boot/ssh`
Prepare wifi with `/boot/wpa_supplicant`
```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=CA

network={
 ssid="name"
 psk="pass"
}
```
Boot pi and follow prerequisites (update, dist-upgrade, reboot, set wifi country).
Install using quick install `curl -sL https://install.raspap.com | bash -s -- --yes`
*Do not reboot*
Open admin interface and confirm wifi client page can scan successfully
Set up hotspot, save, start, wait for hotspot to start.
Go to wifi client page and confirm scanning no longer works

**Expected behavior**
Wifi Client page should be able to scan for nearby networks and allow user to connect to one.

**Actual behavior**
Wifi Client page is empty, not possible to select another network.

**Additional context**
As mentioned above, running `sudo wpa_cli -i wlan0 scan_results` will show nearby networks despite the Wifi Client page being empty.

I may get a Pi Zero W in the future and I'll be able to test on that hardware at that time.

alexFab, 2020-09-02:
i have exactly the same issue on a pi 4 /  RaspberryPiOS Lite.

billz, 2020-09-02:
@mdeabreu which kernel version do you have?

I'm unable to repo on an RPi 4 / RPiOS lite, although scanning disables the AP (see below).
`Linux raspberrypi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux`

Performing a scan with wpa_cli while in AP-STA returns a list of wifi stations in the UI (identical to running the cli from the shell), but doing so causes the AP to disassociate from wlan0. Relevant syslog entries:

```
Sep  2 19:34:16 raspberrypi kernel: [936200.192217] ieee80211 phy1: brcmf_cfg80211_stop_ap: setting AP mode failed -52
Sep  2 19:34:16 raspberrypi kernel: [936200.192716] ieee80211 phy1: brcmf_link_down: WLC_DISASSOC failed (-52)
Sep  2 19:34:16 raspberrypi dhcpcd[21000]: wlan0: carrier lost
Sep  2 19:34:16 raspberrypi dhcpcd[21000]: wlan0: deleting address 2a01:e34:ef85:fbf0:ca62:45b8:cf37:1f26/64
Sep  2 19:34:16 raspberrypi avahi-daemon[363]: Withdrawing address record for 2a01:e34:ef85:fbf0:ca62:45b8:cf37:1f26 on wlan0.
```
tl;dr: stop hostapd before running wpa_cli. connect your client, then restart the AP.

alexFab, 2020-09-02:
`sudo wpa_cli -i wlan0 scan`
`sudo wpa_cli -i wlan0 scan_result`

works fine

but if is press Rescan (with or without running ap),  syslog:
```
lighttpd[1432]: We trust you have received the usual lecture from the local System
lighttpd[1432]: Administrator. It usually boils down to these three things:
lighttpd[1432]:     #1) Respect the privacy of others.
lighttpd[1432]:     #2) Think before you type.
lighttpd[1432]:     #3) With great power comes great responsibility.
lighttpd[1432]: sudo: no tty present and no askpass program specified
```

kernel:
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux
```

closed 2020-11-08 by stale[bot]