RaspAP/raspap-webgui #659: Wrong Password only 1 Device can connect

SchnoppDog, 2020-08-30:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
I've installed RaspAP with the quick installer and made some changes for my home network. Everything is set up I have simultaneous AP and managed mode enabled for other related project reasons. I can connect to the AP fine and there are no issues accessing the internet. The problem rises when I want to connect more than 1 devices to my AP. I get the error "wrong password". I've tested 4 different devices if one is connected to the AP the other 4 devices can't connect. If I disconnect the currently connected device and delete the wifi from my mobile-wifi-options another device can connect to the AP, but the remaining three can't.

I know that there have been some issues related to my problem, but they never provided a solution for my case.

**Your environment**
* Raspberry Pi hardware: Pi 3 Model B+   
* Raspbian version: Newest raspberry pi OS (Buster 10) released 2020-08-20
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install or Manual setup? Quick Install
* Using default configuration? N
* Simultaneous AP and managed mode? Y
* Onboard wireless chipset or external adapter? Onboard chipset 
* Other software or services running with RaspAP? No

**Steps to reproduce**
Downloading and executing the installer. 

- In **Hotspot** I changed AP-modus and Wifi-Client to on. Operating Interface is uap0. 
- In **DHCP-Server** I changed first IP-Address to 192.168.1.2 and last IP to 192.168.1.255. DNS is set to 1.1.1.1 and 8.8.8.8
- In **Network** I changed uap0 IP to 192.168.1.1/24 with a gateway of 192.168.0.1 with no DNS. For wlan0 I changed the IP to 192.168.0.5/24 with a gateway of 192.168.0.1 and no DNS. eth0 is disabled. 

**Expected behavior**
I expect that more than 1 devices can connect to my AP.

**Actual behavior**
Only one client per time can connect to the AP. If one client is connected other clients can't connect to the AP and get an error "wrong password" although passwor is set correctly

**Screenshots**
No screenshots available

**Additional context**
This is driving me crazy and other related issues discussing this problem aren't provided with a helpful solution in my case.

billz, 2020-08-31:
Impossible to determine w/o logs. Enable DHCP and hostapd logging, try connecting a client and look for errors. You can also debug dhcp with `journalctl -xe |grep dhcp`.
Polite reminder that this is not a troubleshooting forum.
There are countless threads with this topic in [these forums](https://github.com/billz/raspap-webgui/wiki/Reporting-issues#issue-policy). You could also try asking at [/r/RaspAP](https://www.reddit.com/r/RaspAP/).

adamsmutnicki, 2020-09-01:
I had exactly same situation. Fresh pi OS, fresh raspap installation. WPA2 +CCMP and only one device allowed. When trying to resolve it I've lost connection to hostapd sms I'm traveling so unable to extract logs.

SchnoppDog, 2020-09-01:
Thank you billz for telling me the options on how to log incoming connections. I will look at the provided links. I am sorry I bothered you didn't know where to post my problem. Have a great day.

billz, 2020-09-01:
@SchnoppDog no worries. I had to draw a line at troubleshooting b/c I wouldn't get anything done on this project otherwise. logging, the FAQ and other community resources exist for this reason. hope this makes sense.

adamsmutnicki, 2020-10-24:
I've done some testing. Seems that the issue of only 1 device able to connect is related to onboard wifi on Pi3B+. It's repeatable. Changing adapter to a popular Edimax one resolved the problem immediately. So, it is not possible (unless someone knows what to change) to created AP based on integrated Wlan adapter.

billz, 2020-10-25:
Test conditions: 3B+, onboard adapter, AP-STA mode. Wireless mode 802.11n, security WPA2 / CCMP. 

```
$ uname -a
Linux raspberrypi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux
```

```
Using interface uap0 with hwaddr dc:▒▒▒▒▒▒▒▒▒▒▒▒▒ and ssid "raspi-webgui"
uap0: interface state COUNTRY_UPDATE->ENABLED
uap0: AP-ENABLED 
uap0: STA 6c:▒▒▒▒▒▒▒▒▒▒▒▒▒ IEEE 802.11: associated
uap0: AP-STA-CONNECTED 6c:▒▒▒▒▒▒▒▒▒▒▒▒▒
uap0: STA 6c:▒▒▒▒▒▒▒▒▒▒▒▒▒ RADIUS: starting accounting session 06A201D1F7B69711
uap0: STA 6c:▒▒▒▒▒▒▒▒▒▒▒▒▒ WPA: pairwise key handshake completed (RSN)
uap0: STA e0:▒▒▒▒▒▒▒▒▒▒▒▒▒ IEEE 802.11: associated
uap0: AP-STA-CONNECTED e0:▒▒▒▒▒▒▒▒▒▒▒▒▒
uap0: STA e0:▒▒▒▒▒▒▒▒▒▒▒▒▒ RADIUS: starting accounting session 71358F1FABC41A99
uap0: STA e0:▒▒▒▒▒▒▒▒▒▒▒▒▒ WPA: pairwise key handshake completed (RSN)
uap0: STA 76:▒▒▒▒▒▒▒▒▒▒▒▒▒ IEEE 802.11: associated
uap0: AP-STA-CONNECTED 76:▒▒▒▒▒▒▒▒▒▒▒▒▒
uap0: STA 76:▒▒▒▒▒▒▒▒▒▒▒▒▒ RADIUS: starting accounting session A2A402BEB771FC13
uap0: STA 76:▒▒▒▒▒▒▒▒▒▒▒▒▒ WPA: pairwise key handshake completed (RSN)
uap0: STA 10:▒▒▒▒▒▒▒▒▒▒▒▒▒ IEEE 802.11: associated
uap0: AP-STA-CONNECTED 10:▒▒▒▒▒▒▒▒▒▒▒▒▒
uap0: STA 10:▒▒▒▒▒▒▒▒▒▒▒▒▒ RADIUS: starting accounting session 723AF1843EB41939
uap0: STA 10:▒▒▒▒▒▒▒▒▒▒▒▒▒ WPA: pairwise key handshake completed (RSN)
```
[image]

Confirmed good internet and (reasonable) throughput for AP-STA mode with 4 simultaneous client connections.

@adamsmutnicki in the absence of dhcp and/or hostapd logs I have to call this unreproducible. Again, use the log facilities stated above, review the issue policy with regard to [default settings](https://github.com/billz/raspap-webgui/wiki/Reporting-issues#default-settings) and search for errors on popular forums.

closed 2020-12-31 by stale[bot]