RaspAP/raspap-webgui #666: wlan1 keeps being disabled

kanecharles92, 2020-09-08:
Hi Guys,

Recently came across `raspap-webgui` and am having issues with a [modmypi wireless adapter](https://thepihut.com/products/wifi-dongle-ultra-long-range-high-gain-w-5dbi-antenna) on my Raspberry Pi 4 Model B.

I keep seeing `wlan1` (the usb wifi adapter) go offline as per:

[image]

The logs are showing:

```
Configuration file: /etc/hostapd/hostapd.conf
wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
Using interface wlan1 with hwaddr 00:0f:00:10:10:63 and ssid "G-WAGON"
wlan1: interface state COUNTRY_UPDATE->ENABLED
wlan1: AP-ENABLED 
wlan1: STA 24:62:ab:fd:24:34 IEEE 802.11: authenticated
wlan1: STA 24:62:ab:fd:24:34 IEEE 802.11: associated (aid 1)
wlan1: AP-STA-CONNECTED 24:62:ab:fd:24:34
wlan1: STA 24:62:ab:fd:24:34 RADIUS: starting accounting session ACFFD6D085DA53FF
wlan1: STA 24:62:ab:fd:24:34 WPA: pairwise key handshake completed (RSN)
wlan1: interface state ENABLED->DISABLED
wlan1: AP-STA-DISCONNECTED 24:62:ab:fd:24:34
wlan1: AP-DISABLED 
wlan1: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Configuration file: /etc/hostapd/hostapd.conf
wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
Using interface wlan1 with hwaddr 00:0f:00:10:10:63 and ssid "G-WAGON"
wlan1: interface state COUNTRY_UPDATE->ENABLED
wlan1: AP-ENABLED 
wlan1: STA 24:62:ab:fd:24:34 IEEE 802.11: disassociated
wlan1: STA 24:62:ab:fd:24:34 IEEE 802.11: authenticated
wlan1: STA 24:62:ab:fd:24:34 IEEE 802.11: associated (aid 1)
wlan1: AP-STA-CONNECTED 24:62:ab:fd:24:34
wlan1: STA 24:62:ab:fd:24:34 RADIUS: starting accounting session 4EF6566098440806
wlan1: STA 24:62:ab:fd:24:34 WPA: pairwise key handshake completed (RSN)
wlan1: interface state ENABLED->DISABLED
wlan1: AP-STA-DISCONNECTED 24:62:ab:fd:24:34
wlan1: AP-DISABLED 
wlan1: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Configuration file: /etc/hostapd/hostapd.conf
wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
Using interface wlan1 with hwaddr 00:0f:00:10:10:63 and ssid "G-WAGON"
wlan1: interface state COUNTRY_UPDATE->ENABLED
wlan1: AP-ENABLED 
wlan1: STA 24:62:ab:fd:24:34 IEEE 802.11: authenticated
wlan1: STA 24:62:ab:fd:24:34 IEEE 802.11: associated (aid 1)
wlan1: AP-STA-CONNECTED 24:62:ab:fd:24:34
wlan1: STA 24:62:ab:fd:24:34 RADIUS: starting accounting session F625079D2038744C
wlan1: STA 24:62:ab:fd:24:34 WPA: pairwise key handshake completed (RSN)
wlan1: interface state ENABLED->DISABLED
wlan1: AP-STA-DISCONNECTED 24:62:ab:fd:24:34
wlan1: AP-DISABLED 
wlan1: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Configuration file: /etc/hostapd/hostapd.conf
wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
Using interface wlan1 with hwaddr 00:0f:00:10:10:63 and ssid "G-WAGON"
wlan1: interface state COUNTRY_UPDATE->ENABLED
wlan1: AP-ENABLED 
wlan1: STA 24:62:ab:fd:24:34 IEEE 802.11: authenticated
wlan1: STA 24:62:ab:fd:24:34 IEEE 802.11: associated (aid 1)
wlan1: AP-STA-CONNECTED 24:62:ab:fd:24:34
wlan1: STA 24:62:ab:fd:24:34 RADIUS: starting accounting session 1D33A64C8502CE9A
wlan1: STA 24:62:ab:fd:24:34 WPA: pairwise key handshake completed (RSN)
wlan1: AP-STA-DISCONNECTED 24:62:ab:fd:24:34
Failed to set beacon parameters
wlan1: INTERFACE-DISABLED 
[... 259 lines cut ...]
wlan1: INTERFACE-DISABLED 
wlan1: INTERFACE-ENABLED 
Failed to set beacon parameters
wlan1: INTERFACE-DISABLED 
wlan1: INTERFACE-ENABLED 
Failed to set beacon parameters
wlan1: STA b4:fb:e4:2b:85:1c IEEE 802.11: disassociated due to inactivity
wlan1: STA b4:fb:e4:2b:85:1c IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
wlan1: STA b4:fb:e4:2b:85:1c IEEE 802.11: disassociated
wlan1: INTERFACE-DISABLED 
```

If I restart the hotspot it seems to work for a period of time and then die?

The interface itself looks like it's disappearing from the pi itself entirely:

[image]

I've not had this adapter for long, is it possible the adapter is faulty or that I need a powered usb hub?

I'm using an official raspi power adapter to power the pi if that makes any difference.

Unsure if this is raspap attempting to configure the wrong parameters which is causing the `wlan1` interface to die or whether it could be something else. Would greatly appreciate some advice on where to look further.

Cheers!

kanecharles92, 2020-09-08:
I've noticed that `/etc/hostapd/hostapd.conf` is populated as follows:

```
driver=nl80211
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
auth_algs=1
wpa_key_mgmt=WPA-PSK
beacon_int=100
ssid=G-WAGON
channel=1
hw_mode=g
ieee80211n=0
wpa_passphrase=REDACTED
interface=wlan1
wpa=2
wpa_pairwise=CCMP
country_code=AU
ignore_broadcast_ssid=0
```

[My adapter is a RT2800 chipset](https://thepihut.com/products/wifi-dongle-ultra-long-range-high-gain-w-5dbi-antenna), is `nl80211` the correct parameter for `driver`?

billz, 2020-09-09:
The driver setting is correct for your chipset. 

> possible the adapter is faulty or that I need a powered usb hub?
 
Check syslog for power related issues. You can also check iwconfig and try disabling power management on wlan1. I've done zero testing with Ralink adapters, so perhaps others can chime in.

kanecharles92, 2020-09-09:
It appears that power management for wlan1 is off:

```
pi@raspberrypi:~ $ iwconfig
eth0      no wireless extensions.

br-90795360eed3  no wireless extensions.

br-31f211893829  no wireless extensions.

br-74fef0ee24ad  no wireless extensions.

wlan0     IEEE 802.11  ESSID:off/any
          Mode:Managed  Access Point: Not-Associated   Tx-Power=31 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on

wlan1     IEEE 802.11  ESSID:off/any
          Mode:Managed  Access Point: Not-Associated   Tx-Power=36 dBm
          Retry short  long limit:2   RTS thr:off   Fragment thr:off
          Power Management:off

docker0   no wireless extensions.

lo        no wireless extensions.
```

I've commented out the `beacon_interval` value for now to see if that makes any difference as the errors looked to be when it was attempting to set that value. It seems more stable so far, will report back in a day or two to see it that has fixed it.

Is that something that can be exposed as a configurable option within raspap so that it doesn't configure `beacon_interval=100` every time I save the hotspot settings?

billz, 2020-09-10:
It's a good idea to expose `beacon_int` as an advanced option. 
Boolean config settings such as `disassoc_low_ack` have also helped some users, so this could be added as well.

billz, 2020-09-11:
[image]

kanecharles92, 2020-10-09:
Hi @billz, for reference it *appears* to have been hardware related, see [here](https://unix.stackexchange.com/questions/610338/raspberry-pi-4-model-b-external-rt3070-wifi-adapter-hostapd-crashing).

Thanks heaps!

billz, 2020-10-15:
@kanecharles92 good to know, thanks for sharing your findings.

closed 2020-10-15 by billz