RaspAP/raspap-webgui #669: Feature request: Custom URL filters

darcyrush, 2020-09-14:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [ ] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the feature**
It would be nice to have a dedicated tab where we can paste our own URLs to block.

**Your environment**
* Raspberry Pi 4
* Raspberry Pi OS Lite Buster 2020-08-20
* Followed the project prerequisites? (Y)
* Checked the project FAQ? (Y)
* RaspAP Quick Install
* Using default configuration? (N)
* Simultaneous AP and managed mode? (N)
* Onboard wireless chipset

**Expected behavior**
To see another tab in the Ad blocking section with a text block and save button. Save button saves all entries to a separate file `/etc/raspap/adblock/custom.txt` with `0.0.0.0` prepended, stop/starts dnsmasq which always reads the file regardless.

**Actual behavior**
I need to edit `/etc/hosts` then stop and start `dnsmasq`

**Additional context**
I've been looking for an easy-to-setup and maintained project which I can use to block URLs before encrypting my traffic using OpenVPN and **without using DNS services** since my VPN provider intercepts any DNS request which isn't directed to their private encrypted DNS servers.

As a linux user with root access, adding the entries on my own local machine is pointless as it takes seconds to remove them. At least I can change the passwords for RaspAP once its set up and physically hide them. Enough of a deterrent to actually just do my work instead of wasting hours on Reddit.

It doesn't solve Firefoxes DoH option, but thats a different problem to solve.

billz, 2020-09-17:
Just to be clear, implementing this as a Linux hosts file would block entire hosts, not specific URIs. See the existing hostnames.txt as an example.

darcyrush, 2020-09-18:
Apologies, I should have been more clear. Blocking entire hosts would be completely fine.

billz, 2020-09-19:
IP addresses and hosts are validated from user input. Invalid items are indicated with the line number. A valid custom blocklist is added to `090_adblock.conf` with the `addn-hosts` option.

[image]

billz, 2020-09-20:
PR merged. Thanks for the suggestion :)

PR #671 (2020-09-20): Support for Adblock custom blocklists
Resolves #669
closed 2020-09-20 by billz