RaspAP/raspap-webgui #672: Adblocking and hosts file ignored if DHCP DNS Servers defined

darcyrush, 2020-09-20:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
Perhaps I am misunderstanding the setup of the DHCP Server and the wording underneath 'Upstream DNS servers' but when DNS Servers are defined in the Server settings tab, entries in the etc hosts file and adblock hostnames.txt are ignored.

**Your environment**
* Raspberry Pi 4
* Raspberry Pi OS Lite Buster 2020-08-20
* Followed the project prerequisites? (Y)
* Checked the project FAQ? (Y)
* RaspAP Quick Install
* Using default configuration? (Y)
* Simultaneous AP and managed mode? (N)
* Onboard wireless chipset

**Steps to reproduce**
Define DNS Servers 1 and 2 in DHCP server settings (I used 1.0.0.1 and 1.1.1.1)

**Expected behavior**
Entries in `/etc/hosts` and `/etc/raspap/adblock/hostnames.txt` should be taken into account.

**Actual behavior**
The hosts file entries and adblock entries are ignored. 

**Additional context**
Stop starting dnsmasq or adblock doesn't show this issue, a full pi reboot is needed each time.

Reading the hosts and adblock entries works fine when no DNS servers are defined at all (either in Server settings or Advanced - so your ISPs DNS servers are automatically used instead) and when only the upstream DNS servers are set and enabled in Advanced.

billz, 2020-09-21:
You're correct. Defining these DNS servers adds `dhcp-option=6` to `/etc/dnsmasq.d/090_raspap.conf`:

```
dhcp-option=6,1.0.0.1,1.1.1.1
```

which bypasses local name resolution and the ad blocking hosts.  dnsmasq interprets the [special value 0.0.0.0](http://www.thekelleys.org.uk/dnsmasq/docs/dnsmasq-man.html) to mean "the address of the machine running dnsmasq" so you can try:

```
dhcp-option=6,0.0.0.0,1.0.0.1,1.1.1.1
```

In theory this will use dnsmasq as the first nameserver, and the last two as backups (untested).

darcyrush, 2020-09-23:
Ah ok. Thank you for the clarification, I guess this can be closed then.

Perhaps a short text description above the two DNS fields would clarify help clarify this for future users?

billz, 2020-09-24:
Actually, I'm curious if adding the special `0.0.0.0` value worked in this case 😉 
If it queries the local hostnames first, then your ISP's DNS this could be a useful option for the UI.

darcyrush, 2020-09-27:
Let me try this and get back to you.

closed 2020-12-03 by stale[bot]