RaspAP/raspap-webgui #676: Change eth0 route to wwan0

Imcon, 2020-09-22:
I have a Sixhab LTE HAT and I can get the modem all working, ping google.com -I wwan0, etc.

sudo ping google.com -I wwan0
PING  (172.217.9.142) from 10.190.147.55 wwan0: 56(84) bytes of data.
64 bytes from dfw25s26-in-f14.1e100.net (172.217.9.142): icmp_seq=1 ttl=96 time=247 ms
64 bytes from dfw25s26-in-f14.1e100.net (172.217.9.142): icmp_seq=2 ttl=96 time=179 ms
64 bytes from dfw25s26-in-f14.1e100.net (172.217.9.142): icmp_seq=3 ttl=96 time=128 ms

Raspap works and loads but I can't figure out how to route all wlan0 traffic to wwan0, basically to have a LTE hotspot with a Pi attached. Been working on it for weeks and every time I read some about how it should work "setting the basic settings interface to wwan0", etc. ANY help from someone that actually has this setup and it's working for them would be greatly appreciated!
Thanks, Dan

Interface | IP Address | Gateway | Internet Access
-- | -- | -- | --
No route to the internet found

billz, 2020-09-23:
Have you tried asking at the Sixfab forums? https://community.sixfab.com/

Imcon, 2020-09-23:
Yes, and they say if the modem connects and you can ping/curl a site it’s doing it’s job.
All that I need to know is how to get Raspap to set wwan0 as the Internet connection and then everything would route through that interface to the net. Wlan0 is still the wifi access point but the incoming and outgoing from that interface would go through Raspap.
I’d think this would be the same as having multiple interface cards on a device and routing the internet connection to one of them rather than always eth0 or wlan0 right?

Dan Horn - CTO
Chief Technology Officer
Imcon International Inc.
2874 Johnson Ferry Road
Suite 150
Marietta, GA 30062

billz, 2020-09-23:
The second wireless interface is set with `iw dev`: 
https://github.com/billz/raspap-webgui/blob/a7f9afdd7aafb235e97acc96d48a01cd00b77c86/includes/wifi_functions.php#L148

eg., 
```
phy#0
	Interface wlan1
		ifindex 4
		wdev 0x1
		addr 80:░░░░░░░░░
		ssid raspi-webgui
		type AP
		channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
		txpower 20.00 dBm
phy#1
	Interface wlan0
		ifindex 3
		wdev 0x100000001
		addr dc:░░░░░░░░░
		ssid Freebox-Router
		type managed
		channel 104 (5520 MHz), width: 40 MHz, center1: 5510 MHz
		txpower 31.00 dBm
```

which then provides the internet connection ...

[image]

Probably your LTE hat is exposed as a third interface so it isn't being assigned (default is next available).
Try unbinding `wlan0` if it isn't in use or force `wwan0` in `getWifiInterface()`.

Imcon, 2020-09-23:
Billz,

Really appreciate your help with this so I’m trying to give you as much info as I can and not bug you too much.

Goal = Have a wifi hotspot with RaspAp that shows a public accessible SSID that get’s it’s internet connection through a SixFab LTE Hat card on my Raspberry Pi.
Using RaspAp as the hotspot software, Dietpi as the OS (Debian stripped), and a Pi 4.

My assumption is that wlan0 is being used as the wifi connection to the LTE HAT so changing HotSpot->Basic->Interface to wwan0 means the wifi hotspot is somehow supposed to be active while at the same time it’s routing in and out through the LTE connection?

Needless to say I need some hand holding here but I think with the $100 price point of these SixHab Hat’s that an option that “just works” once the modem is live would a great promoter of Raspap?

For what it’s worth I’ve got this running, as hotspot through my home wifi router (Jack Sparrow) connection and ethernet, under Dietpi.

RaspAp would be a tremendous add-on to the Dietpi optimized software interface.

Thank you for your assistance and I look forward to hearing from you soon.

wlan0 = wifi hotspot interface
wwan0 = LTE based internet connection
eth0 = not plugged in

I actually don’t see anything in the Internet Connection screen per below.

[cid:6DEA57C8-1665-4CAD-9880-6E41EF94D855@attlocal.net]
[cid:3315B783-317E-45E5-A06F-356C6CDDCF91@attlocal.net]
[cid:47B4F2A2-32AA-4397-821B-6CF1D091F8F2@attlocal.net]
[cid:FAD36063-49D9-4FB4-8E11-BA422153591C@attlocal.net]

root@dietpi:/home/dietpi# route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         10.190.147.56   0.0.0.0         UG    0      0        0 wwan0
10.190.147.48   0.0.0.0         255.255.255.240 U     0      0        0 wwan0
192.168.1.0     0.0.0.0         255.255.255.0   U     0      0        0 wlan0

[cid:image001.png@01D471F8.0CBC92A0]

Dan Horn - CTO
Chief Technology Officer
Imcon International™ Inc.
2874 Johnson Ferry Rd, Suite 100
Marietta, GA 30062
www.imconintl.com<http://www.imconintl.com/>
linkedin.com/in/danhorn<http://linkedin.com/in/danhorn>
Connecting the Planet

©2020 Imcon International, Inc.. All Rights Reserved. This communication, as well as any and all attachments to this communication, constitutes an electronic communication within the meaning of the Electronic Communications Privacy Act, 18 USC 2510, and its disclosure is strictly limited to the recipient intended by the sender of this message.  This communication and attachment may contain confidential and privileged material for the sole use of the intended recipient and receipt by anyone other than the intended recipient does not constitute a loss of the confidential or privileged nature of the communication. Any review or distribution by others is strictly prohibited. IF YOU ARE NOT THE INTENDED RECIPIENT DO NOT OPEN THE ATTACHMENTS. Please contact the sender by return electronic mail and delete all copies of this communication.

https://github.com/billz/raspap-webgui/blob/a7f9afdd7aafb235e97acc96d48a01cd00b77c86/includes/wifi_functions.php#L148

[networking]<https://user-images.githubusercontent.com/229399/94033850-a8290f00-fdc1-11ea-93d9-d5a549ada238.png>

billz, 2020-09-23:
Dan, feel free to drop me an email. Address is in index.php

Budman1758, 2020-09-28:
This is the exact setup that I am trying to get working also.  My SixFab LTE modem is connecting and has a data connection and is working as its supposed to. 

@Imcon did you get this working yet?  If so I would love to hear how you got it working.  My use case is a carputer headunit setup. I have a gps signal feeding a navigation app and want to use the lte connection as a hotspot for another computer in the car and for other clients to connect.

I keep going to the wiki [here](https://github.com/billz/raspap-webgui/wiki/FAQs#interfaces)

This would seem to be what is needed but I never get past the first step which is to "bind" the 2 interfaces.  Cannot find a setting for that so I assume it requires some config file magic to do that. And if so I have no idea how to make that happen. I have no problem editing config files but need the particulars.

TIA!!

Imcon, 2020-10-01:
So far no luck, I keep trying to figure out how to make the wwan0 interface the primary internet connection but can’t seem to make that happen.

Will try to keep you in the loop but I really think this is a Raspap issue and I’m not talented enough to rebuild the source.

Let me know if you’ve managed to get it working Carl.

Dan Horn
+1.770.337.1667
dan.horn@me.com
LinkedIn: http://linkedin.com/in/danhorn <http://linkedin.com/in/danhorn>

“Remember, there’s a fine line between people with vision and those that are just seeing things.” - Mike Morton

darnokb, 2020-10-02:
@Imcon, first of all install ifmetric
then ifmetric wwan0 50
you're done.

Budman1758, 2020-10-03:
@darnokb I tried your suggestion and still cannot get it to work. Do you have this exact configuration set up and working?  If so I would really like to see your config details.

Imcon, 2020-10-04:
I still have gotten it working but will try the ifmetric install this week.

Dan Horn - CTO
Chief Technology Officer
Imcon International Inc.
2874 Johnson Ferry Road
Suite 150
Marietta, GA 30062

darnokb, 2020-10-05:
@Budman1758, when new interface is connected to the raspap (running on Armbian) I log in via ssh and just type command
`ifmetric <interface> <value>`
remember that lowest value means first in routing table

@Imcon, in case of question feel free to ask

please remember that @billz implemented already one feature suggested by myself, which could solve your problems: https://github.com/billz/raspap-webgui/issues/686

There is also a way to set metric permanently in /etc/dhcpd.conf (raspberry pi os) or /etc/network/interfaces (armbian) - did not tested this yet

billz, 2020-10-05:
#687 sets metric values in /etc/dhcpcd.conf so they are persistent.

Imcon, 2020-10-05:
Caveats:

1) It took me weeks to configure something that should have taken minutes, PLEASE SixFab and Raspap make this a simple use case instead of a nightmare!

2) This still doesn’t work after reboot, this is because I’m a script kitty and not a linux guru, any and all advise on solving this more elegantly would be GREATLY appreciated.

3) This worked for me, your milage may vary and I personally can’t help, this was primarily a trial and error process that worked by some stroke of the imagination and my stubbornness not to send the LTE card back to Sixfab.

4) From my experience one of the BIG problems is SixFab’s website telling you to change the Hotspot interface to wwan0, this doesn’t work if you actually want a HOTSPOT since the wlan0 interface is used for the actual wifi HOTSPOT…sigh

5) My LTE carrier is ATT using a KORE Wireless sim and their APN, be sure to check if you get an IP once the LTE card is up, if not then you don’t have their software working. Move the Pi to someplace that gets a signal is most likely the problem as these aren’t big antennas on the LTE card.
________________

Ok all, here’s what I found out and how I made it work even though a reboot sets it all back to not working again.

Wrote a Raspian Lite image to a 32gb SD card with ApplePi Backer (Macintosh)
Installed Raspap
curl -sL  https://install.raspap.com | bash

****>>>>> BIG BIG NOTE <<<<<***

Sixfab scripts are hardwired to use a host directories of “pi” if you try to install these with different host directories you will fail, you have been warned.
(Next try is to get this mess working on DietPi to keep the OS size to a minimum, film at eleven.)

*> On with the show <*

Installed Sixfab QMI software
./qmi_install.sh

Installed SixFab QMI software auto connect on boot
./install_auto_connect.sh

**> Note, I had to install auto connect install a couple of times since sometime the file size was zero but it still completed
Check if the file size is zero (0) bytes and if so reinstall
Should look like this:
root@raspberrypi:/home/pi# ls -al /usr/src
total 24
drwxr-xr-x  5 root root 4096 Oct  5 15:44 .
drwxr-xr-x 10 root root 4096 Aug 20 06:26 ..
drwxr-xr-x 23 root root 4096 Oct  5 12:29 linux-headers-5.4.51+
drwxr-xr-x 23 root root 4096 Oct  5 12:29 linux-headers-5.4.51-v7+
drwxr-xr-x 23 root root 4096 Oct  5 12:29 linux-headers-5.4.51-v7l+
-rw-rw-rw-  1 root root  229 Oct  5 15:44 qmi_reconnect.sh

Reloaded all daemons
systemctl daemon-reload

Made sure QMI software was running
systemctl start qmi_reconnect.service

Enabled it to run on reboot (redundant to auto_connect.sh but what the hell)
systemctl enable qmi_reconnect.service

**> This is the kicker, disable dchpcd as it steps on or is stepped on by Busybox’s udhcpcd service! **
systemctl disable dhcpcd.service

**> This is a biggie as well, get rid of the eth0 interface if you aren’t using it (my usecase)
I’m a complete newbie to using routing tables and eth0 alway over rode the wwan0 internet access so I add wwan0 as a default

route del default eth0
route add default wwan0

Should look something like this:
root@raspberrypi:/home/pi# route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         0.0.0.0         0.0.0.0         U     0      0        0 wwan0
default         __IP Address__  0.0.0.0         UG    0      0        0 wwan0
__IP Address__  0.0.0.0         255.255.255.0   U     304    0        0 wlan0
__IP Address__  0.0.0.0         255.255.255.240 U     0      0        0 wwan0
[... 30 lines cut ...]

root@raspberrypi:/home/pi# cat /etc/network/interfaces
# interfaces(5) file used by ifup(8) and ifdown(8)

# Please note that this file is written to be used with dhcpcd
# For static IP, consult /etc/dhcpcd.conf and 'man dhcpcd.conf'

# Include files from /etc/network/interfaces.d:
source-directory /etc/network/interfaces.d
root@raspberrypi:/home/pi#

[cid:3F6F3172-687E-4B02-831F-B661C5457D72][cid:507AB502-8925-402D-AA07-E819ED1D49D0][cid:E277AD06-D36C-4D5F-B8C1-D54E15C1E8EA][cid:E9DD81FD-EA21-4410-987A-4B6012318229][cid:D8C01CAB-F17B-4EE1-9A62-7D998902B2E8][cid:70C84380-5235-4BD7-BBA7-8797D0005AF3][cid:F7D27A5E-304F-44A4-9B80-8E75FB4AFAD1][cid:9ECBD290-4BB3-4A15-8A55-8D15CE5AD337][cid:6DD7E2C7-9936-4260-B3A6-284BE09B1DB6]

[cid:image001.png@01D471F8.0CBC92A0]

Dan Horn - CTO
Chief Technology Officer
Imcon International™ Inc.
2874 Johnson Ferry Rd, Suite 100
Marietta, GA 30062
www.imconintl.com<http://www.imconintl.com/>
linkedin.com/in/danhorn<http://linkedin.com/in/danhorn>
Connecting the Planet

©2020 Imcon International, Inc.. All Rights Reserved. This communication, as well as any and all attachments to this communication, constitutes an electronic communication within the meaning of the Electronic Communications Privacy Act, 18 USC 2510, and its disclosure is strictly limited to the recipient intended by the sender of this message.  This communication and attachment may contain confidential and privileged material for the sole use of the intended recipient and receipt by anyone other than the intended recipient does not constitute a loss of the confidential or privileged nature of the communication. Any review or distribution by others is strictly prohibited. IF YOU ARE NOT THE INTENDED RECIPIENT DO NOT OPEN THE ATTACHMENTS. Please contact the sender by return electronic mail and delete all copies of this communication.

closed 2020-10-06 by billz

billz, 2020-10-06:
@Imcon RaspAP is free software. It has no 'out-of-the-box' support for LTE hats beyond the existing Networking UIs. If you'd like to see this functionality added, please consider [sponsoring this project](https://github.com/billz/raspap-webgui#support-us) at one of our commercial tiers. Thanks & best of luck with your implementation.

maha-haji, 2021-08-03:
> This is the exact setup that I am trying to get working also. My SixFab LTE modem is connecting and has a data connection and is working as its supposed to.
>
> @Imcon did you get this working yet? If [...]

@Budman1758 did you ever get this going?