RaspAP/raspap-webgui #680: WiFi Client page distorted when only one SSID

RobertDuncan42, 2020-09-23:
* [x] This is a bug report
* [x ] I searched existing issues before opening this one
* [x ] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
Incorrect output on WiFi Client page when only one SSID is displayed and the screen is larger than a phone. The one displayed card is squished horizontally. Noticed this on a desktop while testing, then confirmed on a tablet, display is ok on a phone (Samsung S10+).

**Your environment**
* Raspberry Pi hardware: Pi 3 Model B+  
* Raspbian version: examples: Buster Lite
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install ? Y
* Using default configuration? Y
* Simultaneous AP and managed mode? N
* Onboard wireless chipset or external adapter? Alfa AWUS036NHA, RT3070 
* Other software or services running with RaspAP? N

**Steps to reproduce**
Get into a position with only one SSID in range (I removed the antenna on my dongle). Then open WiFi Client page and rescan.

**Expected behavior**
Card of the single SSID should look like it does with multiply SSID.

**Actual behavior**
The single SSID card is squished, label fields overflow and truncate.

**Screenshots**
Attached desktop screen capture.

**Additional context**
..

[image]

billz, 2020-09-23:
@RobertDuncan42 looks like a bug, thanks for reporting.

billz, 2020-09-23:
[image]

commit 054c5383 (2020-09-23): Wifi client layout fix. Resolves #680
closed 2020-09-23 by billz