RaspAP/raspap-webgui #681: Handling of different clients (WLAN, mobile data modem/router etc)

zbchristian, 2020-09-26:
I implemented different clients in RaspAP, including a corresponding status display on the dashboard (see below). The clients include mobile data modem/router, network via usb tethering.
[image]
[image]
[image]

The fork at https://github.com/zbchristian/raspap-webgui contains a set of shell scripts to handle the connection and to extract the status of different devices. Mobile devices are currently limited to Huawei E1550 (3G modem) and E3372h (4G mobile router stick).

The installer has been updated to install all of the required packages, config files, service files and udev rules. So, this fork should install out of the box.  

The current version still requires some manual configuration for the modem. In the file `/etc/wvdial.conf` the PIN of the SIM and the APN has to be added. For the mobile router the PIN should be disabled and web gui should have no login. These restrictions will be removed soon.

@billz mainly the dashboard is changed in RaspAP. But these changes are substantial. A script tries to find out which client is active and the display is modified accordingly. For this I implemented a client type number (0-5). All informations about the client is collected by a script (getClients.sh) and returned in json format. These data are used for the display. 
This version is still missing GUI components for the configuration of the mobilde data devices and configurations for more devices.
I would be happy to create a pull request, just drop me a line.

zbchristian, 2020-09-29:
Update: the fork contains now a GUI to enter the PIN and APN settings.

billz, 2020-09-29:
@zbchristian looks great. I'll carve out some time and dive into it. thanks

zbchristian, 2020-11-29:
Another update: modified client configuration via json file (client types, udev rules). UI changed accordingly. User can now assign a client type to a device. 
Modified version in branch Full-UDEV-Client-Config of https://github.com/zbchristian/raspap-webgui

billz, 2021-01-29:
@zbchristian ran an install from your Full-UDEV-Client-Config feature branch. I have only checked the new UIs, which look great, and still need to test the client device connectivity. There's an impressive amount of work done here. 

There are some conflicts with the v2.6 release, but nothing not too difficult to resolve. 

Any suggestions for testing the various supported clients? (I may need to procure some new hardware)

zbchristian, 2021-01-29:
@billz I tested with Android phones in thethering mode, but not yet with an iPhone. Its actually not easy to recognize these. An Android device uses the "rndis_host" driver and for an iPhone I hope, that "iPhone" or "Apple" shows up in the device or vendor name . 
The WLAN sticks should be no problem, but a driver is always required (sometimes tricky with Realtek based devices). 

Concerning mobile usb devices: most devices sold in Germany, even from ISPs like Telekom, are Huawei devices (e.g. E3272). These come in two flavours: dialin modem (AT commands) or Hilink (e.g. E3272h-320). The first is handled by wvdial and the latter by a HTTP API (my script: onoff_huawei_hilink.sh). 

I own the mentioned E3272h-320 and an older 3G dialin modem E1550. I did all the tests with these two devices. 

It would be good to test more devices, especially the no-name types which are available for 15-20€ on Ebay. These are often router sticks, which include WLAN and a 4G modem. I assume, that these  devices just show up as an ethernet port (e.g. eth1). Some comments hint at a powering problem (get very hot), which might be a problem for a raspberry pi. The information about the SIM card status/PIN, switch off/on the connection depends on the availablility of an API (via HTTP). If no API exists, one has to accept it as an ethernet port, which can not be switched on/off etc.

billz, 2021-01-29:
Super, thanks. I've ordered a E3272h-320 for testing. In the meantime, I'll swap a SIM into an Android phone and give it a shot. It appears that USB tethering is indeed known to be problematic on [iOS 14 devices](https://forum.openwrt.org/t/has-the-ios14-tethering-issue-been-resolved/74949).

The refactoring done with the 2.6-beta branch moved all of the DHCP functions formerly under Networking to (surprise) the DHCP Server UI. The upshot of this is the former now has space for your new tabs. I also think the functionality is grouped more logically this way, ie., "Network devices" under Networking.

billz, 2021-01-30:
USB tethering indicated nicely in the dashboard (installed with `zbchristian/master` branch).

[image]

I assume `zbchristian/Full-UDEV-Client-Config` is meant to be merged with your current master branch, correct? I did an install with this feature branch and writing udev rules to `/etc/udev/rules.d/80-raspap-net-devices.rules` works, however the client was not displayed on the dashboard. Still navigating the code deltas.

> A script tries to find out which client is active and the display is modified accordingly. For this I implemented a client type number (0-5)

Got it, works great. I follow the logic of `getClients.sh` (well documented!). Anticipating the end user, I imagine the first request will be to ability to override this from the UI. I suppose this could be done with a 'Set default' button or similar on the **Network devices** tab?

zbchristian, 2021-01-30:
@billz getClients.sh should no longer be needed and is actually removed in the branch Full-UDEV-Client-Config. I moved the code to includes/get_clients.php . The configuration of clients types and templates for UDEV rules is stored in client_udev_prototypes.json . This is no longer based on type numbers, but by name "eth", "wlan", "ppp", "hilink", "phone" .
Hope that makes sense!

jimmont, 2021-03-04:
USB tethering with iOS 14 (in my case iPhone 6s iOS 14.4) is fixed. Note **_bridged network traffic appears not to work at all with iOS**_ and when this is attempted results in link-local addresses for eth1 (169.254.../fe80::..., I'm using aarch64 raspberrypi os 5.10.x), however routed traffic does and obviously is how macbooks share their wifi network connection as well (and how I was able to figure this out). Also note **_this addressing won't change_** (in my experience) until the iOS device network settings are reset (Settings > General > Reset > [Reset Network Settings]).

billz, 2021-05-01:
Docs for this feature are here https://docs.raspap.com/net-devices/
Thanks everyone for your input.

closed 2021-05-01 by billz