RaspAP/raspap-webgui #682: resolve raspap default ip address into mydomain.com or mydomain/

ojon1st, 2020-09-26:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [] This is a bug report
* [x ] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
First this is not a bug but a further configuration option.
I want to have the right configuration to turn 10.3.141.1 into home.com or home/

**Your environment**
* Raspberry Pi 4
* Raspbian Buster Desktop 64 bits
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install or Manual setup?
* Using default configuration? Y
* Simultaneous AP and managed mode? Y
* Onboard wireless chipset  
* Other software or services running with RaspAP? No

**Expected behavior**
Get the right dnsmasq and hosts configuration to access to Rpi from "home/" or "home.com" instead of 10.3.141.1

billz, 2020-09-26:
Seems redundant to the existing avahi-daemon / mDNS implementation of `raspberrypi.local`.

ojon1st, 2020-09-26:
> Seems redundant to the existing avahi-daemon / mDNS implementation of `raspberrypi.local`.

Thank you for your quick answer.
This works well when accessing from the raspberrypi browser.
But when accessing from another device / computer, it's not working.
I only can access from http://10.3.141.1

billz, 2020-09-26:
I think assigning hostnames is best left to the user, thanks

closed 2020-09-26 by billz