RaspAP/raspap-webgui #683: Armbian cannot run wifi client

darnokb, 2020-09-26:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [X] I searched existing issues before opening this one
* [X] I checked the FAQ before creating this issue
* [X] I have read and understand the issue reporting guidelines 

**Describe the bug**
Not able to be a wifi client as wlan0 nor wlan1
Cannot connect to wifi

root@nanopi-r1:~# wpa_supplicant -B -Dnl80211,wext -c/etc/wpa_supplicant/wpa_sup                                                                             plicant.conf -iwlan0
Successfully initialized wpa_supplicant
nl80211: kernel reports: Authentication algorithm number required
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
ctrl_iface exists and seems to be in use - cannot override it
Delete '/var/run/wpa_supplicant/wlan0' manually if it is not used anymore
Failed to initialize control interface 'DIR=/var/run/wpa_supplicant GROUP=netdev                                                                             '.
You may have another wpa_supplicant process already running or the file was
left by an unclean termination of wpa_supplicant in which case you will need
to manually remove this file before starting wpa_supplicant again.

nl80211: deinit ifname=wlan0 disabled_11b_rates=0

**Your environment**
* NanoPi R1
* Armbian Buster
* Followed the project prerequisites? (Y)
* Checked the project FAQ? (Y)
* RaspAP Quick Install or Manual setup? Quick
* Using default configuration? (Y)
* Simultaneous AP and managed mode? (N)
* Onboard wireless chipset or external adapter?  Both
* Other software or services running with RaspAP? N

**Steps to reproduce**
Install RaspAP on Armbian Buster and try to configure it as wifi client

**Expected behavior**
Connect to wifi network via wlan0 or wlan1

**Actual behavior**
root@nanopi-r1:~# wpa_supplicant -B -Dnl80211,wext -c/etc/wpa_supplicant/wpa_sup                                                                             plicant.conf -iwlan0
Successfully initialized wpa_supplicant
nl80211: kernel reports: Authentication algorithm number required
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
ctrl_iface exists and seems to be in use - cannot override it
Delete '/var/run/wpa_supplicant/wlan0' manually if it is not used anymore
Failed to initialize control interface 'DIR=/var/run/wpa_supplicant GROUP=netdev                                                                             '.
You may have another wpa_supplicant process already running or the file was
left by an unclean termination of wpa_supplicant in which case you will need
to manually remove this file before starting wpa_supplicant again.

nl80211: deinit ifname=wlan0 disabled_11b_rates=0

**Screenshots**
N/A

**Additional context**
It does not matter if I wlan0 is AP and wlan1 is client, behavior is the same for both wlan* adapters and exactly the same error occurs.

darnokb, 2020-09-26:
Disabling NetworkManager from systemd helped a little - I am able to connect via wpa_cli to wifi, but cannot via web portal.

billz, 2020-09-27:
> You may have another wpa_supplicant process already running or the file was
left by an unclean termination of wpa_supplicant in which case you will need
to manually remove this file before starting wpa_supplicant again.

Relevant [FAQ](https://github.com/billz/raspap-webgui/wiki/FAQs#scanning).

darnokb, 2020-09-29:
@billz, my fault, you are right.
I've disabled NetworkManager and changed wifi client from wlan1 to wlan0, problem solved.

closed 2020-09-29 by darnokb

billz, 2020-09-29:
@darnokb good to know, thanks for sharing your solution.