RaspAP/raspap-webgui #685: WiFi Disconnects randomly and fails authentification

grea09, 2020-09-29:
* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
The WiFi on the RaspberryPi 3 seems to drop randomly. The issue arrises under a few minutes after a client connects to the AP. All clients gets kicked and cannot reconnect because of an authentication issue. Restarting hostapd fixes the issue for a few minutes again.

**Your environment**
* Raspberry Pi 3 Model B+
* Raspbian version Buster Lite
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install
* Using default configuration? Y (Except OpenVPN)
* Simultaneous AP and managed mode? N
* Onboard wireless chipset
* Other software or services running with RaspAP? Docker, HomeAssistant

**Steps to reproduce**
Headless install using ethernet. Install everything for RaspAP then install Docker and HomeAssistant. Setup AP with custom SSID and password. Update iptable using the code under the FAQ for Docker and make it persistent using `iptables-manager` and `iptables-save`. Disable powersave.

**Expected behavior**
Stable WiFi access point.

**Actual behavior**
Randomly dropping access point

**Additional context**
Logs:

```
root@Ananas:~# iwconfig
...
wlan0     IEEE 802.11  Mode:Master  Tx-Power=31 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
```

```
root@Ananas:~# cat /etc/hostapd/hostapd.conf 
driver=nl80211
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
auth_algs=1
wpa_key_mgmt=WPA-PSK 
beacon_int=100
ssid=Secret
channel=1
hw_mode=g
ieee80211n=0
ieee80211w=0
wpa_passphrase=secret
interface=wlan0
wpa=2
wpa_pairwise=TKIP CCMP
country_code=FR
ignore_broadcast_ssid=0
```

```
sept. 29 09:05:33 Ananas dhcpcd[27681]: wlan0: carrier acquired
sept. 29 09:05:33 Ananas dhcpcd[27681]: wlan0: IAID f3:94:81:de
sept. 29 09:05:33 Ananas dhcpcd[27681]: wlan0: adding address fe80::8581:ee2:642f:f630
sept. 29 09:05:33 Ananas avahi-daemon[21940]: Registering new address record for fe80::8581:ee2:642f:f630 on wlan0.*.
sept. 29 09:05:33 Ananas dhcpcd[27681]: wlan0: probing address 10.3.141.1/24
sept. 29 09:05:33 Ananas wpa_supplicant[344]: wlan0: Failed to get BSSID
sept. 29 09:05:33 Ananas hostapd[3185]: wlan0: STA 34:ea:34:f5:25:03 IEEE 802.11: associated
sept. 29 09:05:33 Ananas wpa_supplicant[344]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
sept. 29 09:05:33 Ananas dhcpcd[27681]: wlan0: carrier lost
sept. 29 09:05:33 Ananas dhcpcd[27681]: wlan0: deleting address fe80::8581:ee2:642f:f630
sept. 29 09:05:33 Ananas hostapd[3185]: wlan0: STA 34:ea:34:f5:25:03 RADIUS: starting accounting session 3E79007555FCA133
sept. 29 09:05:33 Ananas hostapd[3185]: wlan0: STA 34:ea:34:f5:25:03 WPA: pairwise key handshake completed (RSN)
sept. 29 09:05:33 Ananas avahi-daemon[21940]: Withdrawing address record for fe80::8581:ee2:642f:f630 on wlan0.
sept. 29 09:05:33 Ananas dhcpcd[27681]: wlan0: carrier acquired
sept. 29 09:05:33 Ananas dhcpcd[27681]: wlan0: IAID f3:94:81:de
sept. 29 09:05:33 Ananas dhcpcd[27681]: wlan0: adding address fe80::8581:ee2:642f:f630
sept. 29 09:05:33 Ananas avahi-daemon[21940]: Registering new address record for fe80::8581:ee2:642f:f630 on wlan0.*.
sept. 29 09:05:33 Ananas dhcpcd[27681]: wlan0: probing address 10.3.141.1/24
sept. 29 09:05:33 Ananas dhcpcd[27681]: wlan0: carrier lost
sept. 29 09:05:33 Ananas dhcpcd[27681]: wlan0: deleting address fe80::8581:ee2:642f:f630
sept. 29 09:05:33 Ananas avahi-daemon[21940]: Withdrawing address record for fe80::8581:ee2:642f:f630 on wlan0.
sept. 29 09:05:33 Ananas dhcpcd[27681]: wlan0: carrier acquired
sept. 29 09:05:33 Ananas dhcpcd[27681]: wlan0: IAID f3:94:81:de
sept. 29 09:05:33 Ananas dhcpcd[27681]: wlan0: adding address fe80::8581:ee2:642f:f630
sept. 29 09:05:33 Ananas avahi-daemon[21940]: Registering new address record for fe80::8581:ee2:642f:f630 on wlan0.*.
sept. 29 09:05:33 Ananas dhcpcd[27681]: wlan0: probing address 10.3.141.1/24
sept. 29 09:05:39 Ananas dhcpcd[27681]: wlan0: using static address 10.3.141.1/24
sept. 29 09:05:39 Ananas avahi-daemon[21940]: Joining mDNS multicast group on interface wlan0.IPv4 with address 10.3.141.1.
sept. 29 09:05:39 Ananas avahi-daemon[21940]: New relevant interface wlan0.IPv4 for mDNS.
sept. 29 09:05:39 Ananas avahi-daemon[21940]: Registering new address record for 10.3.141.1 on wlan0.IPv4.
sept. 29 09:05:39 Ananas dhcpcd[27681]: wlan0: adding route to 10.3.141.0/24
```

It seems to be related to `carrier lost` where the interface `wlan0` gets reset or something.

billz, 2020-09-29:
Please read this if you skipped it the first time. Thanks
https://github.com/billz/raspap-webgui/wiki/Reporting-issues

closed 2020-09-29 by billz

JamesxX, 2024-09-16:
Having this exact problem and this page showed up on Google. Very glad that instead of a solution I have a link to report problems instead