RaspAP/raspap-webgui #689: No stats while using wlan1

Ewwwer, 2020-10-05:
* [x] This is a bug report
* [ ] I searched existing issues before opening this one
* [ ] I checked the FAQ before creating this issue
* [ ] I have read and understand the issue reporting guidelines 

**Describe the bug**
Hi, i'm using an external wifi antenna (Realtek RTL8188EU) on Raspberry Pi 4b 4gb. 
It works fine but I have noticed that the consumption statistics are only available for eth0 and wlan0. Do I have it wrongly configured?
How can I have consumption statistics also in wlan1?
Thank you
[image]
[image]
[image]

billz, 2020-10-05:
Execute `vnstat -i wlan1` and post the result.

Ewwwer, 2020-10-06:
Thanks @billz, the report says "unble to read database /var/lib/vnstat/wlan1: Document not found.
But the strange thing is that I was broadcasting the signal through wlan1

billz, 2020-10-06:
The error indicates a problem with vnstat. You can try a couple of things:

1. vnstat may not be able to write to `/var/lib/vnstat`. check which user the daemon is running under with `ps aux | grep vnstatd`. if it's inactive or running under an account other than vnstat, give the service a kick with `sudo systemctl restart vnstat.service`
2. force an update for wlan1 with `sudo vnstat -u -i wlan1`. this will create the database for that interface if it doesn't exist

Ewwwer, 2020-10-06:
Hi, I’ve tried all the possibilities that you’ve told me, this is the result: 
[image]

But for the other way, the things are changed for the command vnstat -i wlan1: 
[image]

I’m gonna use it and see if your suggestions tips are working well. Thanks a lot

Ewwwer, 2020-10-06:
Good news, it works perfectly! Thanks a lot 
[image]

billz, 2020-10-06:
great! glad to hear it

brunopiras, 2020-10-16:
Same problem, but on command line appears that vnstat works, on ubuntu 20.04 (not raspberry):

apu2@routerx:~$ ps aux | grep vnstatd
vnstat      1138  0.0  0.0   5608  3768 ?        Ss   13:58   0:02 /usr/sbin/vnstatd -n
apu2        3848  0.0  0.0  11084   728 pts/1    S+   16:16   0:00 grep --color=auto vnstatd

vnstat -i wlp4s0
Database updated: 2020-10-16 16:15:00

   wlp4s0 since 2020-10-15

          rx:  64,05 MB      tx:  496,57 MB      total:  560,62 MB

   monthly
                     rx      |     tx      |    total    |   avg. rate
     ------------------------+-------------+-------------+---------------
       2020-10      64,05 MB |   496,57 MB |   560,62 MB |    3,31 kbit/s
     ------------------------+-------------+-------------+---------------
     estimated     125,88 MB |   980,29 MB |     1,11 GB |

   daily
                     rx      |     tx      |    total    |   avg. rate
     ------------------------+-------------+-------------+---------------
     yesterday      31,37 MB |   260,08 MB |   291,44 MB |   26,98 kbit/s
         today      32,68 MB |   236,49 MB |   269,17 MB |   36,81 kbit/s
     ------------------------+-------------+-------------+---------------
     estimated      48,26 MB |   349,28 MB |   397,54 MB |

[image]

billz, 2020-10-16:
@brunopiras Ubuntu 20.04 is untested and unsupported at this time.

brunopiras, 2020-10-18:
> @brunopiras Ubuntu 20.04 is untested and unsupported at this time.

Thanks for your reply
I would like to help with this problem, the new version of vnstat generates in the /var/lib/vnstat folder a single database with the name "vnstat.db" that contains all the information about the network interfaces while, the old versions of vnstat generated a database for each interface that was detected with the name of the interface itself.

I solved this by uninstalling version 2.2.x of vnstat:

sudo apt-get purge vnstat 

 download and installing an old one (I used the 1.14):

sudo dpkg -i vnstat_1.14-1ubuntu2_amd64.deb

then blacklisting the program to avoid future updates:

echo "vnstat hold" | sudo dpkg --set-selections

Now works well..
Thanks

[image]

closed 2020-12-24 by stale[bot]