RaspAP/raspap-webgui #690: Quick Installer opeerrors

tweak42, 2020-10-06:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [ ] I searched existing issues before opening this one
* [ ] I checked the FAQ before creating this issue
* [ ] I have read and understand the issue reporting guidelines 

**Describe the bug**
A clear and concise description of what the bug is.

RaspAP Install: Configure OpenVPN support
Install OpenVPN and enable client configuration? [Y/n]: 
RaspAP Install: Installing OpenVPN and enabling client configuration
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  easy-rsa libccid liblzo2-2 libpkcs11-helper1 opensc opensc-pkcs11 pcscd
Suggested packages:
  pcmciautils openvpn-systemd-resolved
The following NEW packages will be installed:
  easy-rsa libccid liblzo2-2 libpkcs11-helper1 opensc opensc-pkcs11 openvpn pcscd
0 upgraded, 8 newly installed, 0 to remove and 0 not upgraded.
Need to get 1,958 kB of archives.
After this operation, 5,437 kB of additional disk space will be used.
Do you want to continue? [Y/n] Abort.
[ ✘ error ]  Unable to install openvpn 
Enabling openvpn-client service on boot
Failed to enable unit: Unit file openvpn-client@client.service does not exist.
[ ✘ error ]  Unable to enable openvpn-client daemon 
RaspAP Install: Creating OpenVPN control scripts
changed ownership of '/etc/raspap/openvpn/configauth.sh' from root:root to root:www-data
[ ✓ ok ] 
RaspAP Install: Adding raspap.sudoers to /etc/sudoers.d/090_raspap
RaspAP Install: Symlinked wpa_supplicant hooks for multiple wlan interfaces
RaspAP Install: Unmasking and enabling hostapd service
Removed /etc/systemd/system/hostapd.service.
Synchronizing state of hostapd.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable hostapd
[ ✓ ok ] 
RaspAP Install: Installation completed
This project needs your help! Please consider supporting RaspAP on Open Collective or GitHub:

> https://opencollective.com/raspap
> https://github.com/sponsors/billz

The system needs to be rebooted as a final step. Reboot now? [y/N]: 

**Your environment**
* Raspberry Pi hardware (examples: Pi 3 Model B+, Pi Zero W)  
* Raspbian version (examples: Buster Lite, Buster Desktop)
* Followed the project prerequisites? (Y/N)
* Checked the project FAQ? (Y/N)
* RaspAP Quick Install or Manual setup?
* Using default configuration? (Y/N)
* Simultaneous AP and managed mode? (Y/N)
* Onboard wireless chipset or external adapter? 
* Other software or services running with RaspAP?

**Steps to reproduce**
Tell us how to reproduce this issue. Provide as much detailed information as possible.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Actual behavior**
Tell us what you observed instead.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

billz, 2020-10-07:
> Do you want to continue? [Y/n] Abort.
[ ✘ error ] Unable to install openvpn

Did you intentionally abort the install? This prompt is from apt-get, not the quick installer.

tweak42, 2020-10-07:
Did not abort.  I just press enter at each y/n prompt.  From "Install OpenVPN and enable client configuration? [Y/n]:" it automatically continues until prompt "The system needs to be rebooted as a final step. Reboot now? [y/N]:"

billz, 2020-10-07:
```
RaspAP Install: Configure OpenVPN support
Install OpenVPN and enable client configuration? [Y/n]: 
RaspAP Install: Installing OpenVPN and enabling client configuration
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Suggested packages:
  openvpn-systemd-resolved
The following NEW packages will be installed:
  openvpn
0 upgraded, 1 newly installed, 0 to remove and 32 not upgraded.
Need to get 0 B/426 kB of archives.
After this operation, 1,019 kB of additional disk space will be used.
Preconfiguring packages ...
Selecting previously unselected package openvpn.
(Reading database ... 109107 files and directories currently installed.)
Preparing to unpack .../openvpn_2.4.7-1_armhf.deb ...
Unpacking openvpn (2.4.7-1) ...
Setting up openvpn (2.4.7-1) ...
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for systemd (241-7~deb10u4+rpi1) ...
Enabling openvpn-client service on boot
```

possible you had a network error during install?

tweak42, 2020-10-07:
I was able to repeat the openvpn error 3 times using clean raspi os lite images to sd cards.  The later 2 times re-running the quick installer after reboot in the last step and the openvpn did download and install as expected.  

I did not try re-running the script after NOT rebooting at the last step.  Possibly there might be some sort of dns resolving issue cause by my network adblocker, I'll disable adblocker and try another clean sd card flash and install when I get some time.

robmcc1983, 2020-10-08:
I had the same problem 2 days ago, clean install using quick install script failed to install openvpn and openvpn client daemon.

I tried running script again but still no luck, I installed the missing packages manually and then open vpn would work. 
Hope this helps

robmcc1983, 2020-10-08:
Just tried again on fresh install, this is what i get.

RaspAP Install: Configure OpenVPN support
Install OpenVPN and enable client configuration? [Y/n]: y
RaspAP Install: Installing OpenVPN and enabling client configuration
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  easy-rsa libccid liblzo2-2 libpkcs11-helper1 opensc opensc-pkcs11 pcscd
Suggested packages:
  pcmciautils openvpn-systemd-resolved
The following NEW packages will be installed:
  easy-rsa libccid liblzo2-2 libpkcs11-helper1 opensc opensc-pkcs11 openvpn
  pcscd
0 upgraded, 8 newly installed, 0 to remove and 0 not upgraded.
Need to get 1,958 kB of archives.
After this operation, 5,437 kB of additional disk space will be used.
Do you want to continue? [Y/n] Abort.
[ ✘ error ]  Unable to install openvpn
Enabling openvpn-client service on boot
Failed to enable unit: Unit file openvpn-client@client.service does not exist.
[ ✘ error ]  Unable to enable openvpn-client daemon
RaspAP Install: Creating OpenVPN control scripts
changed ownership of '/etc/raspap/openvpn/configauth.sh' from root:root to root:www-data
[ ✓ ok ]
RaspAP Install: Adding raspap.sudoers to /etc/sudoers.d/090_raspap
RaspAP Install: Symlinked wpa_supplicant hooks for multiple wlan interfaces
RaspAP Install: Unmasking and enabling hostapd service
Removed /etc/systemd/system/hostapd.service.
Synchronizing state of hostapd.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable hostapd
[ ✓ ok ]
RaspAP Install: Installation completed
This project needs your help! Please consider supporting RaspAP on Open Collective or GitHub:

> https://opencollective.com/raspap
> https://github.com/sponsors/billz

The system needs to be rebooted as a final step. Reboot now? [y/N]:

billz, 2020-10-10:
Did you follow the [prerequisites](https://github.com/billz/raspap-webgui#prerequisites) before installing RaspAP?

tweak42, 2020-10-13:
Ok after doing a clean image and rerunning the quick installer script more times than I can remember.
Entering nothing and uppercase Y at the install openvpn Y/n prompt fails with the abort error. N and n work as expected.
Is the script calling the 'apt-get install -y openvpn' command interpreted correctly?

This post implies that apt aborts when it can't read stdin.
https://askubuntu.com/questions/509852/why-does-apt-get-abort-by-itself-as-though-id-pressed-n

billz, 2020-10-14:
This was a weird one. Finally able to repro and resolved. Thanks

closed 2020-10-14 by billz

ochen1, 2020-11-01:
I had this bug a year ago. It's still here now lol

The installer needs to call apt with the `-y` flag! (╯°□°）╯︵ ┻━┻

If I install the packages manually **with the -y flag**, will everything work normally?

ochen1, 2020-11-02:
What I am saying is: Can someone please add a `-y` to this line?
https://github.com/billz/raspap-webgui/blob/master/installers/common.sh#L269

Or should I make a pull request for it? :laughing: its only two characters

Please do something about it! :pleading_face:

ochen1, 2020-11-02:
Wow, I am so patient. I waited a whole 3min 30 seconds before submitting the pull request myself lol

Hopefully the PR will fix it, and one less error in the installer brings a thousand more users! :smile:

commit 39dcb506 (2020-10-14): Resolves #690