RaspAP/raspap-webgui #692: No way to provide DHCP IP to eth0

pierrot10, 2020-10-16:
Good day, 
I successfully configure my Raspbery Pi3 with RaspAP. I can connect with the WiFi with my laptop.
However, when I connect another Pi from the eth0 poert, no IP address is provided by the first Pi.
Additonaly, the first Pi with RaspAP has the following address 
```
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST> mtu 1500
inet 169.254.216.121 netmask 255.255.0.0 broadcast 169.254.255.255
```

Is a issu or the is a specific was to provide IP address thrus the eth0 and wlan0?

My goal is to provide internet acccess to the second Pi which is connected to the first Pi with RaspAP

Many thanks for your help.

billz, 2020-10-16:
Duplicate of #324

billz, 2020-11-06:
I've reopened the above issue. Let's track progress there. Closing this one, thanks.

closed 2020-11-06 by billz