RaspAP/raspap-webgui #693: Internet connection lost after starting hotspot in bridged mode

mwelsch, 2020-10-17:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
A clear and concise description of what the bug is.

**Your environment**

* Raspberry Pi hardware (examples: Pi 3 Model B+, Pi Zero W): Raspberry Pi 4 Model B 
* Raspbian version (examples: Buster Lite, Buster Desktop): Buster Desktop (Installed via NOOBS)
* Followed the project prerequisites? (Y/N): No (I'm running Desktop and prerequisits states that only lite is supported)
* Checked the project FAQ? (Y/N): y
* RaspAP Quick Install or Manual setup?: Quick install 
* Using default configuration? (Y/N): n
* Simultaneous AP and managed mode? (Y/N): n
* Onboard wireless chipset or external adapter?: onboard
* Other software or services running with RaspAP?: docker, pivpn

**Steps to reproduce**
Tell us how to reproduce this issue. Provide as much detailed information as possible.

I have following Hotspot settings:

```
Interface:
wlan0
Wireless Mode:
802.11n - 2.4 GHz
Channel:
1
Bridged AP mode:
Yes
Logfile output:
Yes (only activated after things started not working, output is not helpful)
Beacon interval:
100
Country Code:
Austria
```

Internet is provided over `eth0`, using `192.168.1.71` as static IP (configured by pivpn)

**Expected behavior**
I expect raspap to start a hotsport with the given SSID and configured security settings, not managing DHCP/DNS or acting as NAT when pressing the `Start Hotspot` button in the WebUI. I do not expect the IP address of the raspberry to change when turning on Wifi

**Actual behavior**
The connection to my raspberry breaks, because it gets a new ip address assigned. The MAC seems to change, so my Router (at `192.168.1.1`) assigns a new ip to the raspberry.

Clients can connect to the raspberry, however they do not receive an ip address, nor do they have internet connection when setting the ip address manually. (Tested with my Windows 10 PC, Android 8 Tablet and Android 10 Smartphone) 

**Screenshots**

[image]

[image]

[image]

billz, 2020-10-18:
See [this wiki page](https://github.com/billz/raspap-webgui/wiki/Bridged-AP-mode#bridged-ap-mode-limitations) about Bridged AP mode limitations and VPN clients. Your Pi's IP is changing because your upstream router is the DHCP server. For this reason, the static IP assigned by PiVPN is ignored.

Dommel, 2020-11-06:
I Can confirm the Issue. I had a working Configuration in Bridge Mode on Raspberrypi , but after upgrading the Raspian Kernel to 5.4 it is not working anymore even with a fresh Install on a Clean Raspian. But when Flashing an Image of the System before the Kernel Upgrade back it works.

billz, 2020-11-07:
```
$ uname -a
Linux raspberrypi 5.4.72-v7l+ #1356 SMP Thu Oct 22 13:57:51 BST 2020 armv7l GNU/Linux
```

Bridged AP mode enabled, connected several clients, internet speed test nominal, no errors.

```
[303454.267184] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[303454.479006] br0: port 1(eth0) entered blocking state
[303454.479014] br0: port 1(eth0) entered disabled state
[303454.479220] device eth0 entered promiscuous mode
[303454.950611] rtl8192cu: MAC auto ON okay!
[303454.976768] rtl8192cu: Tx queue select: 0x05
[303455.596938] br0: port 2(wlan0) entered blocking state
[303455.596946] br0: port 2(wlan0) entered disabled state
[303455.597095] device wlan0 entered promiscuous mode
[303456.658500] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[303458.777502] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[303459.005124] br0: port 2(wlan0) entered blocking state
[303459.005136] br0: port 2(wlan0) entered forwarding state
[303459.005278] br0: port 1(eth0) entered blocking state
[303459.005296] br0: port 1(eth0) entered forwarding state
[303459.005617] IPv6: ADDRCONF(NETDEV_CHANGE): br0: link becomes ready
[303464.167421] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
```
@Dommel logs or other diagnostic info would be helpful.

Dommel, 2020-11-07:
Okay, thanks for your  fast answer, which Log exactly do you need? The Syslog? Or is there any specific Logfile?

billz, 2020-11-07:
Above was a quick look at dmesg. syslog captures a bit more detail w/ other services, etc.

bringing `br0` up:
```
Nov  7 07:45:11 raspberrypi dhcpcd[2208]: br0: waiting for carrier
Nov  7 07:45:11 raspberrypi dhcpcd[2208]: br0: carrier acquired
Nov  7 07:45:11 raspberrypi kernel: [303459.005124] br0: port 2(wlan0) entered blocking state
Nov  7 07:45:11 raspberrypi kernel: [303459.005136] br0: port 2(wlan0) entered forwarding state
Nov  7 07:45:11 raspberrypi kernel: [303459.005278] br0: port 1(eth0) entered blocking state
Nov  7 07:45:11 raspberrypi kernel: [303459.005296] br0: port 1(eth0) entered forwarding state
Nov  7 07:45:11 raspberrypi kernel: [303459.005617] IPv6: ADDRCONF(NETDEV_CHANGE): br0: link becomes ready
Nov  7 07:45:11 raspberrypi systemd-networkd[2175]: br0: Gained carrier
Nov  7 07:45:11 raspberrypi dhcpcd[2208]: br0: IAID 77:93:19:3f
Nov  7 07:45:11 raspberrypi dhcpcd[2208]: br0: adding address fe80::9956:1f48:48ce:7978
Nov  7 07:45:11 raspberrypi avahi-daemon[327]: Joining mDNS multicast group on interface br0.IPv6 with address fe80::9956:1f48:48ce:7978.
Nov  7 07:45:11 raspberrypi avahi-daemon[327]: New relevant interface br0.IPv6 for mDNS.
Nov  7 07:45:11 raspberrypi avahi-daemon[327]: Registering new address record for fe80::9956:1f48:48ce:7978 on br0.*.
Nov  7 07:45:11 raspberrypi dhcpcd[2208]: br0: soliciting a DHCP lease
Nov  7 07:45:11 raspberrypi dhcpcd[2208]: br0: offered 192.168.1.7 from 192.168.1.254
Nov  7 07:45:12 raspberrypi dhcpcd[2208]: br0: soliciting an IPv6 router
Nov  7 07:45:13 raspberrypi systemd-networkd[2175]: br0: Gained IPv6LL
Nov  7 07:45:13 raspberrypi systemd-timesyncd[283]: Network configuration changed, trying to establish connection.
Nov  7 07:45:13 raspberrypi dhcpcd[2208]: br0: Router Advertisement from fe80::224:d4ff:fea0:52e6
Nov  7 07:45:13 raspberrypi dhcpcd[2208]: br0: adding address 2a01:e34:ef85:fbf0:a2ef:78c7:72af:123d/64
Nov  7 07:45:13 raspberrypi avahi-daemon[327]: Leaving mDNS multicast group on interface br0.IPv6 with address fe80::9956:1f48:48ce:7978.
Nov  7 07:45:13 raspberrypi avahi-daemon[327]: Joining mDNS multicast group on interface br0.IPv6 with address 2a01:e34:ef85:fbf0:a2ef:78c7:72af:123d.
Nov  7 07:45:13 raspberrypi avahi-daemon[327]: Registering new address record for 2a01:e34:ef85:fbf0:a2ef:78c7:72af:123d on br0.*.
Nov  7 07:45:13 raspberrypi avahi-daemon[327]: Withdrawing address record for fe80::9956:1f48:48ce:7978 on br0.
Nov  7 07:45:13 raspberrypi dhcpcd[2208]: br0: adding route to 2a01:e34:ef85:fbf0::/64
Nov  7 07:45:13 raspberrypi dhcpcd[2208]: br0: fe80::224:d4ff:fea0:52e6 is reachable again
Nov  7 07:45:13 raspberrypi dhcpcd[2208]: br0: adding default route via fe80::224:d4ff:fea0:52e6
Nov  7 07:45:13 raspberrypi dhcpcd[2208]: br0: probing address 192.168.1.7/24
```

and connecting a client (iPhone) to the bridged AP on `wlan0`:
```
Nov  7 07:49:36 raspberrypi dnsmasq-dhcp[2919]: DHCPDISCOVER(wlan0) c6:▒▒▒▒▒▒▒▒▒▒▒▒▒
Nov  7 07:49:36 raspberrypi dnsmasq-dhcp[2919]: DHCPOFFER(wlan0) 192.168.0.163 c6:▒▒▒▒▒▒▒▒▒▒▒▒▒
Nov  7 07:49:37 raspberrypi dnsmasq-dhcp[2919]: DHCPREQUEST(wlan0) 192.168.0.163 c6:▒▒▒▒▒▒▒▒▒▒▒▒▒ 
Nov  7 07:49:37 raspberrypi dnsmasq-dhcp[2919]: DHCPACK(wlan0) 192.168.0.163 c6:▒▒▒▒▒▒▒▒▒▒▒▒▒ iPhone
```

Dommel, 2020-11-08:
Okay here is dmesg before Update: 
`Linux raspberrypi 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.118-v7+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1311 SMP Mon Apr 27 14:21:24 BST 2020
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3ac00000
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   Normal zone: 2133 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 242688 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xac/0x4b4 with crng_init=0
[    0.000000] percpu: Embedded 16 pages/cpu s36864 r8192 d20480 u65536
[    0.000000] pcpu-alloc: s36864 r8192 d20480 u65536 alloc=16*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=tty1 root=PARTUUID=769112a9-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 939064K/970752K available (8192K kernel code, 656K rwdata, 2232K rodata, 1024K init, 831K bss, 23496K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
                   lowmem  : 0x80000000 - 0xbb400000   ( 948 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 657 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 832 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 27093 entries in 80 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000023] Switching to timer-based delay loop, resolution 52ns
[    0.000276] Console: colour dummy device 80x30
[    0.000901] console [tty1] enabled
[    0.000955] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000998] pid_max: default: 32768 minimum: 301
[    0.001363] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001399] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002353] CPU: Testing write buffer coherency: ok
[    0.002834] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.003514] Setting up static identity map for 0x100000 - 0x10003c
[    0.003683] rcu: Hierarchical SRCU implementation.
[    0.004508] smp: Bringing up secondary CPUs ...
[    0.005414] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.006386] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.007302] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.007415] smp: Brought up 1 node, 4 CPUs
[    0.007496] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.007521] CPU: All CPU(s) started in HYP mode.
[    0.007541] CPU: Virtualization extensions available.
[    0.008533] devtmpfs: initialized
[    0.021845] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.022119] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.022168] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.022778] pinctrl core: initialized pinctrl subsystem
[    0.023643] NET: Registered protocol family 16
[    0.026770] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.032621] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.032656] hw-breakpoint: maximum watchpoint size is 8 bytes.
[    0.032873] Serial: AMBA PL011 UART driver
[    0.035226] bcm2835-mbox 3f00b880.mailbox: mailbox enabled
[... 583 lines cut ...]
[   18.200929] Bluetooth: Core ver 2.22
[   18.201020] NET: Registered protocol family 31
[   18.201025] Bluetooth: HCI device and connection manager initialized
[   18.201043] Bluetooth: HCI socket layer initialized
[   18.201054] Bluetooth: L2CAP socket layer initialized
[   18.201070] Bluetooth: SCO socket layer initialized
[... cut ...]
[   21.355816] 8021q: adding VLAN 0 to HW filter on device eth0
[   22.139376] br0: port 2(wlan0) entered blocking state
[   22.139387] br0: port 2(wlan0) entered disabled state
[   22.139815] device wlan0 entered promiscuous mode
[   22.345948] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   22.346117] br0: port 2(wlan0) entered blocking state
[   22.346124] br0: port 2(wlan0) entered forwarding state
`
In my readings the Issue is the `lo ate my IP address` . But I have no Idea how to fix that.

Dommel, 2020-11-08:
Okay, here is dmesg before Upgrade 

> Linux raspberrypi 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.118-v7+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1311 SMP Mon Apr 27 14:21:24 BST 2020
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3ac00000
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   Normal zone: 2133 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 242688 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xac/0x4b4 with crng_init=0
[    0.000000] percpu: Embedded 16 pages/cpu s36864 r8192 d20480 u65536
[    0.000000] pcpu-alloc: s36864 r8192 d20480 u65536 alloc=16*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=tty1 root=PARTUUID=769112a9-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 939064K/970752K available (8192K kernel code, 656K rwdata, 2232K rodata, 1024K init, 831K bss, 23496K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
                   lowmem  : 0x80000000 - 0xbb400000   ( 948 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 657 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 832 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 27093 entries in 80 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000023] Switching to timer-based delay loop, resolution 52ns
[    0.000276] Console: colour dummy device 80x30
[    0.000901] console [tty1] enabled
[    0.000955] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000998] pid_max: default: 32768 minimum: 301
[    0.001363] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001399] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002353] CPU: Testing write buffer coherency: ok
[    0.002834] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.003514] Setting up static identity map for 0x100000 - 0x10003c
[    0.003683] rcu: Hierarchical SRCU implementation.
[    0.004508] smp: Bringing up secondary CPUs ...
[    0.005414] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.006386] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.007302] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.007415] smp: Brought up 1 node, 4 CPUs
[    0.007496] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.007521] CPU: All CPU(s) started in HYP mode.
[    0.007541] CPU: Virtualization extensions available.
[    0.008533] devtmpfs: initialized
[    0.021845] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.022119] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.022168] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.022778] pinctrl core: initialized pinctrl subsystem
[    0.023643] NET: Registered protocol family 16
[    0.026770] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.032621] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.032656] hw-breakpoint: maximum watchpoint size is 8 bytes.
[    0.032873] Serial: AMBA PL011 UART driver
[... 585 lines cut ...]
[   18.200929] Bluetooth: Core ver 2.22
[   18.201020] NET: Registered protocol family 31
[   18.201025] Bluetooth: HCI device and connection manager initialized
[   18.201043] Bluetooth: HCI socket layer initialized
[   18.201054] Bluetooth: L2CAP socket layer initialized
[   18.201070] Bluetooth: SCO socket layer initialized
[   18.213954] Bluetooth: HCI UART driver ver 2.3
[... cut ...]
[   21.355816] 8021q: adding VLAN 0 to HW filter on device eth0
[   22.139376] br0: port 2(wlan0) entered blocking state
[   22.139387] br0: port 2(wlan0) entered disabled state
[   22.139815] device wlan0 entered promiscuous mode
[   22.345948] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   22.346117] br0: port 2(wlan0) entered blocking state
[   22.346124] br0: port 2(wlan0) entered forwarding state

In my reading the Problem is "lo ate my IP Adress" but I have no idea how to fix that

billz, 2020-11-08:
@Dommel this is totally unreadable. please edit it with markdown

closed 2020-11-08 by billz

Dommel, 2020-11-08:
> > Log data can be very helpful in context, within reasonable limits.
> 
> @Dommel this is making my eyes melt. I'm unable to repro with the default config. have you modified the [default settings](https://github.com/billz/raspap-webgui/wiki/Reporting-issues#default-settings)? if the answer is 'no', [read this](https://github.com/billz/raspap-webgui/wiki/Reporting-issues#will-raspap-let-me-create-a-configuration-that-breaks-my-hotspot) and create a new issue, filling out the provided template completely.

I realy have no Idea which logfile and in which format I should provide to you, but just in case that someone will run into the same issue, running the script 
`/var/www/html/installers/toggle-bridged-routed.sh `
fixed the issue after reboot