RaspAP/raspap-webgui #694: OpenVPN on RaspAC Leaking DNS Queries

merrickville, 2020-10-20:
I have read the Reporting-issues page.

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**

I am able to use RaspAP with the OpenVPN feature enabled, but the service is leaking DNS queries to my local ISP DNS server.

The OpenVPN config file I upload to RaspAP is attached as a zip file: [nordvpn-uk.zip](https://github.com/billz/raspap-webgui/files/5410957/nordvpn-uk.zip)

**Your environment**
* Raspberry Pi 4 Model B Rev 1.2 (2G RAM)
* Buster Lite, fully updated.
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Install: Quick Install
* Using default configuration? Y
* Simultaneous AP and managed mode? N (Not on same chip)
* Onboard wireless chipset or external adapter: [USB 2.0 Wireless Dongle](https://www.buyapi.ca/product/long-range-300mbps-wireless-n-adapter/) (wlan1) USB Wi-Fi dongle works out of the box on default Buster Lite OS install.
* Other software or services running with RaspAP? None

**Steps to reproduce**

RaspAP standard config, but using Wi-Fi dongle as AP:

  wlan0 = onboard Wi-Fi (STA) connected to my home Wi-Fi
  wlan1 = USB Wi-Fi dongle, hostapd (AP) country CA, channel 3
  Hostapd --> [Advanced] tab: WiFi client AP mode = off

Click OpenVPN link
Enter NordVPN credentials
Upload nordvpn-uk.ovpn (attached zip)
Enable OpenVPN (indicator turns green)
Disconnect laptop Wi-Fi from RaspAP
Connect laptop Wi-Fi to RaspAP

**Expected behavior**

On laptop, visit [https://ipleak.net](https://ipleak.net/)
  Observe United Kingdom/UK IP Address [in my case] (good)
  Observe UK DNS Server (good)

**Actual behavior**

On laptop, visit [https://ipleak.net](https://ipleak.net/)
  Observe UK IP Address (good)
  Observe UK DNS Server (good)
  Observe _local_ ISP DNS server (**bad**)

[https://ipleak.net](https://ipleak.net/) shows my local ISP DNS server _along with_ the remote UK DNS server. Rendering my VPN connection useless for security.

**Screenshots**
[image]

**Additional context**

To solve the leaking DNS issue on Ubuntu 20.04. I added the following option in my *.ovpn file:

   dhcp-option DOMAIN-ROUTE .

On Ubuntu, with the above line added, using the [update-systemd-resolved scripts](https://github.com/jonathanio/update-systemd-resolved) I am able to get all DNS queries to route through the remote DNS server. Expected behavior.

I note on the default installation of RaspAP, 'systemd-resolved' service is disabled. I did NOT try to enable systemd-resolved and test using the update-systemd-resolved scripts. I would like to hear from anyone before I try, I don't want to bork my RaspAP installation.

I had to use the additional following lines on Ubuntu 20.04 OpenVPN service to solve the DNS leak

  script-security 2
  up /etc/openvpn/scripts/update-systemd-resolved
  down /etc/openvpn/scripts/update-systemd-resolved

Note: My home router (sitting between RaspAP & the Internet) uses Cloudflare DoH. But, this does not make any difference, when I use OpenVPN on my Ubuntu laptop without, there are no DNS leaks.

I'm currently using RaspAP (will use after the pandemic) as a secure Travel Router for public Wi-Fi systems.

billz, 2020-10-21:
1. Configure an **Upstream DNS server** in the **DHCP Server** panel.
2. Restart the dnsmasq service.
3. Perform your test again.

merrickville, 2020-10-22:
That's the dhcp-option I was looking for.

Using the web-gui as such:  DHCP Server [Advanced] tab /  [on/off toggle] **Only ever query DNS servers configured below** does indeed put the no-resolv option in the dnsmasq configureation. The only issue is, you need to know the OpenVPN DNS IP to enter it manually.

To automate this process, I wrote a small replacement OpenVPN up/down script. This determines the DNS server assigned to your VPN session, and adds that info in the configuration directory /etc/dnsmasq.d/099-fix-ipleak.conf

Thanks for the point on where to look.

Love your work, this will be on my travel-router (when we can travel again :)

closed 2020-10-22 by merrickville

billz, 2020-10-22:
@merrickville glad to hear it. I'll update the FAQ as I believe this could be useful for others. Thanks for the thorough issue report