RaspAP/raspap-webgui #702: mac address filter ?

defaliz, 2020-11-04:
Hi,
thanks for the good product raspap-webgui
is there a way to filer mac address on hostap ?
i use deny_mac_file=/etc/hostapd/hostapd.deny in hostap.conf that works, but when i restart hostap with webgui my configuration is lost..

regards
Alain

billz, 2020-11-04:
Mac address filtering is not currently supported, although it would be a good advanced setting for hostapd.

defaliz, 2020-11-05:
thanks Billz, 
yes it will be a good idea for the futur version..
regards

pierrepo, 2022-07-21:
This would be very useful indeed!

AgentBlade4, 2022-09-24:
This would be so good because I need it to be open for my DS but I want some security

billz, 2024-09-27:
This has been implemented in #1666 and will be available in the next release.

closed 2024-09-27 by billz

billz, 2024-09-27:
Relevant doc link https://docs.raspap.com/ap-basics/#custom-user-settings