RaspAP/raspap-webgui #703: Cant disable DHCP server

angrynerd, 2020-11-06:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
I am unable to disable the dhcp server. I would like to do this so i can use the AdGuard Home dhcp server instead. I tried doing this by running in bridged mode with the idea that this would let me atleast free up the port and start the docker container for aghome. instead this actually made it to where i couldnt access the pi via the ap network. or via the local address from the router. I would personally prefer if ithere was just a way to disable the dhcp server by its self not switching entirely to bridged mode. 

**Your environment**
* Raspberry Pi hardware: Pi 3b+
* Raspbian version: raspbian lite latest
* Followed the project prerequisites: y
* Checked the project FAQ: y
* RaspAP Quick Install or Manual setup: Quick then configured after.
* Using default configuration: Modified in the menus.
* Simultaneous AP and managed mode: n
* Onboard wireless chipset or external adapter: on board
* Other software or services running with RaspAP: Adguard home however it never actually started.

**Steps to reproduce**
To produce this i have installed raspAP on the os not containerized. I tried to set it to bridge mode as a way of disabling dhcp. did not work and couldnt access it. 

**Expected behavior**
I was hoping that bridged mode would just disable the dhcp server as there is no other way of doing so to my knowledge. 

**Actual behavior**
could no longer access the pi. 

**Additional context**
Really would enjoy this feature or a way to do this manually. Any help with this issue is greatly appreciated.

billz, 2020-11-06:
Stop dhcp with `sudo systemctl stop dhcpcd.service`. You can prevent it from starting on boot with `sudo systemctl disable dhcpcd.service`.

angrynerd, 2020-11-06:
Appreciated. This may be useful in documentation at some point. I will test this when I am back at my desk and then i will close the issue.

angrynerd, 2020-11-09:
Tested this today. No luck i can tell that it stops but when i try and use the port it still says dnsmasq is using it. To my knowledge raspap is the only service on the device using dnsmasq that isnt containerized.

billz, 2020-11-09:
Have you done this? https://github.com/billz/raspap-webgui/wiki/FAQs#adguard

angrynerd, 2020-11-09:
As stated i tried that. But it cause the pi not to be accessable from any interface. I will try and test it again and see if i can get bridged mode to work properly.

angrynerd, 2020-11-09:
Oh my apologies. you linked me to the faq for adguard. yes i have done that its running as the dns server currently. however aghome has several per client features that cant be used in that config becausee it thinks all queries are coming from raspap.

angrynerd, 2020-11-09:
OK i tried bridged mode again but its difficult to setup because you have to enable the aghome dhcp server before the other raspap goes down or the entire pi becomes inaccessable. ironcially even on eth0 rather than the access point. Im gonna try one more time and if it doesnt work then im just gonna use it the way i had it.

angrynerd, 2020-11-09:
It failed again. I cant enable to dhcp service on ag home without the wifi interface being up and it cant be up in bridge mode without becoming inaccessable... I have a feeling there is just something that im missing but for right now i have to just reflash from a backup and try again later. I guess i should probably try and figure out why bridged mode makes it unaccessable rather than using the actual routers dhcp, but im not even sure where to start with that.

billz, 2020-11-09:
The linked FAQ item has been validated in testing and is provided as a courtesy only. Beyond this, there is no support whatsoever for AdGuard home integration with this project. If you find a solution to your issue, you're welcome to share it here.

angrynerd, 2020-11-09:
I appreciate the courtesy an understand the lack of support. I will close this issue for now. I may revisit this issue later but at the moment its not the biggest of deal to go without this functionality. Thank you for your assistence and have a nice day!

closed 2020-11-09 by angrynerd