RaspAP/raspap-webgui #706: No internet Connection eth0 to wlan0

masterdot, 2020-11-08:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [ x] I searched existing issues before opening this one
* [ x] I checked the FAQ before creating this issue
* [x ] I have read and understand the issue reporting guidelines 

**Describe the bug**
I did setup my Pi via this tutorial 
https://3os.org/raspberryPi/TOR-Pi/#torpi_-_raspberry_pi_tor_access_point
eth0 has Connection and wlan0 is working fine too, but there seems to be no route  from eth0 to wlan0. 
I did check the iptables, they don't seem to be the problem. I think it's a small option I did not check. 
Thanks for help in advance! 

**Your environment**
* Raspberry Pi hardware (examples: Pi 3 Model B+, Pi Zero W)  Pi3B
* Raspbian version (examples: Buster Lite, Buster Desktop) Buster lite
* Followed the project prerequisites? (Y/N) y
* Checked the project FAQ? (Y/N) y
* RaspAP Quick Install or Manual setup? Install
* Using default configuration? (Y/N) y
* Simultaneous AP and managed mode? (Y/N)
* Onboard wireless chipset or external adapter? Onboard 
* Other software or services running with RaspAP? Tor

**Steps to reproduce**
Tell us how to reproduce this issue. Provide as much detailed information as possible.

**Expected behavior**
A clear and concise description of what you expected to happen.
Internet connection

**Actual behavior**
Tell us what you observed instead.
No internet connection 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

billz, 2020-11-09:
[image]

masterdot, 2020-11-09:
This does mean to me what? I appreciate your reaction, but I dont get it ;)

billz, 2020-11-09:
You're reporting a bug in this project, for a feature that doesn't exist (TOR), based on a random blog post?

Per the link at the top of this issue:

> If you have installed other software packages on top of RaspAP, particularly those related to networking such as TOR or Pi-hole, please test RaspAP first on a clean install before reporting an issue.

masterdot, 2020-11-09:
I did open this issue, because the AP doesn't work like it should do. 
It should route eth0 to wlan0 to make the internet connection usable. But it doesn't. I did link the tutorial to help look into this problem and why the ap doesn't work like it should. I did remove everything in between like tor and the iptables and it still doesn't route. This is why I'm here and asking for help.
I don't expect you to debug the tutorial, just to help me find why the ap doesn't route.

billz, 2020-11-10:
> It should route eth0 to wlan0 to make the internet connection usable

This project displays routing info on the **Networking** page, but does not detect your custom configuration or modify the routing table for you.

> I don't expect you to debug the tutorial, just to help me find why the ap doesn't route.

This is not a support forum, sorry. Please [read this](https://github.com/billz/raspap-webgui/wiki/Reporting-issues).