RaspAP/raspap-webgui #709: NordVPN Bridged AP mode

fkay0711, 2020-11-09:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
I've installed RaspAP and NordVPN Plugin to set my Raspberry as a VPN Router:
https://vpn-expert.info/vpn-router-raspberry-pi-raspap-and-nordvpn-wi-fi-hotspot-access-point/

It work's until I switch to bridged AP Mode, then I trying to connect with NordVPN I'm receive an Error in the Log Files I see that the AP can't reach the Gateway. I've tried to set the subnet to the same range, without using the Bridged AP Mode, but then I don't have internet access from the access point.

My scenario:
I want to connect my linux sat-receiver to VPN, because of that I bought an raspberry an saw this amazing tool raspap. Though just connect to VPN and then connect the sat-receiver to it.
But I also want to have access from outside my network... Because of that I wanted the same range, Bridged AP Mode would be the best way.. but it don't work. 

I also tried the IP Table commands from the old issues from here, resolve.conf is also correct (my router ip address as gateway) IP Forwarding is also active.

**Just to mention that the step in this tutorial (link below):**
_Important! Now the Raspberry Pi does not work anymore as a hotspot or access point! If you are still connected to the Pi hotspot, you won’t feel a difference. But new connections are impossible — so we need to fix that.
If I want to connect to the Pi Hotspot now, I don’t even get an IP address anymore. That’s why I execute the following two commands and the hotspot assigns me an IP address again and works as designed:_

```
sudo iptables -P INPUT ACCEPT
sudo iptables -P OUTPUT ACCEPT
```
_However, the device does not route my traffic to the Internet. Therefore, the following iptables entries are necessary._

```
sudo iptables -t nat -A POSTROUTING -o tun0 -j MASQUERADE
sudo iptables -A FORWARD -i tun0 -o wlan0 -m state --state RELATED,ESTABLISHED -j ACCEPT
sudo iptables -A FORWARD -i wlan0 -o tun0 -j ACCEPT
```
_Now everything works as designed — but there still is a little catch._ ....

For me that doesn't work I'm still getting no IP address until I disconnect the VPN.

**Your environment**
* Raspberry Pi hardware (Raspberry Pi 4)  
* Raspbian version:
Distributor ID:       Raspbian
Description:	       Raspbian GNU/Linux 10 (buster)
Release:                10
Codename:	       buster

* Followed the project prerequisites? (Y)
* Checked the project FAQ? (Y)
* RaspAP Quick Install or Manual setup? Quick Install
* Using default configuration? (Y) also tried my own
* Simultaneous AP and managed mode? (Y)
* Onboard wireless chipset or external adapter? Onboard
* Other software or services running with RaspAP? NordVPN

**Additional context**
I didn't tried it with OpenVPN oder PiVPN, because there are so many servers from NordVPN and I must download all the .ovpn files and cofingure them, right?

Could you suggest me a solution for my case?
Reach the Raspberry from external and use the AP Mode working with NordVPN?

billz, 2020-11-09:
Relevant wiki link: [Bridged AP mode limitations](https://github.com/billz/raspap-webgui/wiki/Bridged-AP-mode#bridged-ap-mode-limitations). 

> I must download all the .ovpn files and cofingure them, right?

Correct.

fkay0711, 2020-11-10:
The relevant wiki really didn't helped me..
Only how to turn on the bridged mode and how to reach the gui

billz, 2020-11-10:
> Just to mention that the step in this tutorial (link below) also doesn't work for me

This project does not endorse third-party projects that use our software, nor can we offer any support for it. I suggest you post in the comments of that article.

closed 2020-11-10 by billz