RaspAP/raspap-webgui #710: AP+STA on Pi 4 works intermittently

szekelyisz, 2020-11-10:
* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**

When I configure my Raspberry Pi 4 in AP+STA mode, everything is fine on the AP side, but the uplink (STA interface) works intermittently.

**Your environment**
* Raspberry Pi hardware: Pi 4 1GB
* Raspbian version: Raspberry Pi OS 10.6 Buster, all updates installed
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install or Manual setup? Quick install
* Using default configuration? Y
* Simultaneous AP and managed mode? Y
* Onboard wireless chipset
* Other software or services running with RaspAP? None

**Steps to reproduce**

1. Flash a fresh copy of Raspberry Pi OS to SD card
1. Boot the Pi 4
1. Connect the Pi to your WLAN
1. `apt-get update && apt-get dist-upgrade`
1. Make sure Ethernet is disconnected to force install script to use multiple WLAN interfaces
1. Run Quick Install script
1. Enable WiFi client mode on web
1. Connect to the raspi-webgui network
1. Try pinging anything on the Internet either from the Pi or any client connected to it

**Expected behavior**

Ping working without errors.

**Actual behavior**

Here's a sample ping session from the Pi 4 itself:

```
pi@raspberrypi:~ $ ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=10 ttl=116 time=566 ms
64 bytes from 8.8.8.8: icmp_seq=11 ttl=116 time=18.1 ms
64 bytes from 8.8.8.8: icmp_seq=12 ttl=116 time=14.8 ms
64 bytes from 8.8.8.8: icmp_seq=13 ttl=116 time=15.6 ms
64 bytes from 8.8.8.8: icmp_seq=14 ttl=116 time=16.1 ms
64 bytes from 8.8.8.8: icmp_seq=15 ttl=116 time=16.10 ms
64 bytes from 8.8.8.8: icmp_seq=16 ttl=116 time=17.4 ms
64 bytes from 8.8.8.8: icmp_seq=17 ttl=116 time=15.0 ms
64 bytes from 8.8.8.8: icmp_seq=18 ttl=116 time=16.6 ms
64 bytes from 8.8.8.8: icmp_seq=40 ttl=116 time=1421 ms
64 bytes from 8.8.8.8: icmp_seq=41 ttl=116 time=381 ms
64 bytes from 8.8.8.8: icmp_seq=42 ttl=116 time=15.1 ms
64 bytes from 8.8.8.8: icmp_seq=43 ttl=116 time=16.6 ms
64 bytes from 8.8.8.8: icmp_seq=44 ttl=116 time=15.3 ms
64 bytes from 8.8.8.8: icmp_seq=45 ttl=116 time=15.8 ms
64 bytes from 8.8.8.8: icmp_seq=46 ttl=116 time=15.2 ms
64 bytes from 8.8.8.8: icmp_seq=47 ttl=116 time=16.6 ms
64 bytes from 8.8.8.8: icmp_seq=48 ttl=116 time=20.7 ms
64 bytes from 8.8.8.8: icmp_seq=49 ttl=116 time=16.6 ms
64 bytes from 8.8.8.8: icmp_seq=50 ttl=116 time=3069 ms
64 bytes from 8.8.8.8: icmp_seq=51 ttl=116 time=2005 ms
64 bytes from 8.8.8.8: icmp_seq=52 ttl=116 time=965 ms
64 bytes from 8.8.8.8: icmp_seq=53 ttl=116 time=15.9 ms
64 bytes from 8.8.8.8: icmp_seq=54 ttl=116 time=16.8 ms
64 bytes from 8.8.8.8: icmp_seq=55 ttl=116 time=15.4 ms
64 bytes from 8.8.8.8: icmp_seq=56 ttl=116 time=18.7 ms
64 bytes from 8.8.8.8: icmp_seq=57 ttl=116 time=16.4 ms
[... 130 lines cut ...]
64 bytes from 8.8.8.8: icmp_seq=408 ttl=116 time=29.3 ms
64 bytes from 8.8.8.8: icmp_seq=409 ttl=116 time=14.5 ms
64 bytes from 8.8.8.8: icmp_seq=410 ttl=116 time=17.8 ms
[...]
64 bytes from 8.8.8.8: icmp_seq=763 ttl=116 time=15.6 ms
64 bytes from 8.8.8.8: icmp_seq=764 ttl=116 time=15.10 ms
64 bytes from 8.8.8.8: icmp_seq=765 ttl=116 time=16.2 ms
64 bytes from 8.8.8.8: icmp_seq=766 ttl=116 time=16.0 ms
64 bytes from 8.8.8.8: icmp_seq=779 ttl=116 time=884 ms
64 bytes from 8.8.8.8: icmp_seq=780 ttl=116 time=15.8 ms
64 bytes from 8.8.8.8: icmp_seq=781 ttl=116 time=14.6 ms
64 bytes from 8.8.8.8: icmp_seq=782 ttl=116 time=15.2 ms
64 bytes from 8.8.8.8: icmp_seq=783 ttl=116 time=23.6 ms
64 bytes from 8.8.8.8: icmp_seq=784 ttl=116 time=16.8 ms
64 bytes from 8.8.8.8: icmp_seq=785 ttl=116 time=23.0 ms
64 bytes from 8.8.8.8: icmp_seq=786 ttl=116 time=21.0 ms
64 bytes from 8.8.8.8: icmp_seq=787 ttl=116 time=17.3 ms
64 bytes from 8.8.8.8: icmp_seq=788 ttl=116 time=15.7 ms
^C
--- 8.8.8.8 ping statistics ---
791 packets transmitted, 335 received, 57.6485% packet loss, time 840ms
rtt min/avg/max/mdev = 12.696/275.494/3068.982/663.749 ms, pipe 3
```

As you can see from the replies' sequence numbers, the ping stops every 10-30 seconds for a while. The kernel log contains no errors related to the wireless driver.

billz, 2020-11-10:
This project creates an AP-STA configuration for you and nothing more. It makes no promises about the stability or performance of this config. Use of a second wireless adapter will always deliver better results.

szekelyisz, 2020-11-10:
Thank you @billz I understand. I was actually hoping to avoid installing another adapter into the Pi. I basically wanted to see if anybody had this problem before or something's up with Pi 4. I know from the wiki that it works with a Zero W. The hardware says it can do it. 

Maybe this is a kernel driver issue? Maybe channels are wrong? Regulatory issue? According to the hardware information, the channel has to be the same for the STA and AP interface, and uplink APs can change channels. Maybe it switches away, that's why it disappears every now and then? What do you think?

billz, 2020-11-10:
@szekelyisz excellent question. Best place to look for an explanation is probably the Cypress forums. This post on [AP STA concurrency](https://community.cypress.com/community/software-forums/linux/blog/2019/09/30/concurrent-ap-and-sta) largely informed RaspAP's implementation, but I can only speculate to what in brcmfmac might be causing these intermittent delays. Channel switching is a good theory, given that the software AP is constrained by operating on the same channel as the uplink AP.

szekelyisz, 2020-11-12:
Never mind, this really looks like a kernel driver issue. Syslog is flooded with kernel error messages.

closed 2020-11-12 by szekelyisz

delijati, 2021-03-10:
@szekelyisz just out of curiosity ... i have the same problem (3b+ and 4 on buster 2021-01-12) i also tried downgrading/ upgrading the wifi driver. do you still now the kernel driver errors? was it the channel switch thing? mybe my router i confused channel switch. anyway i "fixed" it by using another wifi dongle :/

szekelyisz, 2021-03-10:
Hi @delijati, I didn't investigate further. I put that project on hold, and didn't get back to it since.