RaspAP/raspap-webgui #715: Route hotspot clients through VPN

flaming-gif, 2020-11-16:
I was looking for a solution to route all traffic from clients connected to Wireless AP to a OpenVPN tunnel(tun0). I found a lot of tutorials online where if the Pi is connected to Ethernet this is easily achievable. But I am looking to do the same using only a wifi connection.

Requirement: **(Internet) --[Wifi]-- (Raspberry Pi 3b+) --[OpenVPN tunnel:tun0]--[Wifi-AP]-- (Wifi clients)**

Question: Is this achievable using RaspAP?

I already tried the following, with Wifi connection(no ethernet):
* Hosted Hotspot using RaspAP - Works. I can access internet on connected clients.
* Hosted Hotspot using RaspAP and enabled OpenVPN tunnel - Doesn't work. Connected clients do not get (local) IP address, hence clients do not get internet access. Pi itself can ping internet and myipaddress.com shows VPN IP address.

billz, 2020-11-16:
Welcome to GitHub. Please read this https://github.com/billz/raspap-webgui/wiki/Reporting-issues

Achievable, yes. Currently supported, no. RaspAP's OpenVPN configuration is built around this project's default configuration (eth0 and wlan0) which covers the vast majority of use cases. Traffic is routed through `tun0` with NAT like so:  

https://github.com/billz/raspap-webgui/blob/974c93254e05d5068bffe9d4ffc2df7db91aa5f1/installers/configauth.sh#L33-L39

A purely wireless OpenVPN implementation is a nice-to-have feature, but not on the near term roadmap. PR's are welcome

flaming-gif, 2020-11-16:
Thank you for your reply. I am not fluent with iptables rules, yet. But I can learn about them. Can you in rough words describe how this can be achieved? 

Will just updating iptable rules to route traffic from **uap0** to **tun0** be enough to achieve this? Or do I need to look into some other aspects as well?

billz, 2020-11-16:
Give it a try ;) You could also ask at one the sites mentioned in the link above. This isn't a support forum, sorry.

flaming-gif, 2020-11-17:
:) Thanks for the response. I'll give it a try.

closed 2020-11-17 by flaming-gif