RaspAP/raspap-webgui #718: Add button unclickable in WiFi client config screen for selected network

opentrail, 2020-11-25:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
I am trying to configure a network to connect to using the WiFi client screen. I press Rescan and a list of networks are displayed - great. I select the network I want to connect to. Enter the Passphrase and focus the mouse pointer on the Add button. It is not clickable.

**Your environment**
* Raspberry Pi hardware Pi 400 UK version  
* Raspbian version : Latest version supplied with Pi 400 with update, upgrade and reboot.
* Followed the project prerequisites? (Y)
* Checked the project FAQ? (Y)
* RaspAP Quick Install or Manual setup? Quick install
* Using default configuration? (Y/N)
* Simultaneous AP and managed mode? (Y/N)
* Onboard wireless chipset or external adapter? onboard
* Other software or services running with RaspAP? none

**Steps to reproduce**
Follow steps from Tech Craft https://www.youtube.com/watch?v=YbvSS8MJm2s
Use Quick install
curl -sL https://install.raspap.com | bash
Selected Y for all question except OpenVPN
Followed FAQ as no networks displayed:
sudo wpa_supplicant -B -Dnl80211,wext -c/etc/wpa_supplicant/wpa_supplicant.conf -iwlan0
In browser surf to the IP address and login. Select Wifi Client and Rescan.
Lists a bunch of networks. 
Add Passphrase for the network I want to connect to. Try to click Add but it is unclickable.
Tried Mac Google Chrome and Safari - same problem.

**Expected behavior**
I add the passphrase and the add button is clickable.

**Actual behavior**
Add Passphrase for the network I want to connect to. Try to click Add but it is unclickable.

**Screenshots**
[image]

**Additional context**
Add any other context about the problem here.

opentrail, 2020-11-25:
Noticed my Passphrase is 19 characters. If I continue typing the Add button eventually becomes clickable. I then delete all the characters and re-paste my Passphrase - this time the Add button is clickable and it works.

billz, 2020-11-26:
WPA passphrases must be between 8 and 63 characters. The Add button is disabled until the minimum 8 characters are entered.

billz, 2020-11-26:
https://github.com/billz/raspap-webgui/wiki/FAQs

closed 2021-01-24 by billz