RaspAP/raspap-webgui #720: Failed at removing raspap and trying to find a solution without accessing the headless rpi (no eth no ssh)

boboyoyo7, 2020-11-26:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [ ] I searched existing issues before opening this one
* [ ] I checked the FAQ before creating this issue
* [ ] I have read and understand the issue reporting guidelines 

**Describe the bug**
I want to make my rpi work as a repeater. At first, the connection to my main router was working fine and rpi could connect to the Internet. But the thing is that when I tried to make the hotspot, I messed up and It wasn't working fine . And the big problem is that I couldn't even reconnect to my Wi-Fi from the rpi and so I lost access to my headless rpi because I could only access it by SSH.
So I tried your uninstall script but I failed badly this time not me but the script. At each remove command, the script stopped claiming that there were no such file or directory ... While the file was bel et bien there.

**Your environment**
* Raspberry Pi hardware Pi 3
* Raspbian version atest version supplied with Pi 3
* Followed the project prerequisites? y
* Checked the project FAQ? y
* RaspAP Quick Install or Manual setup? y
* Using default configuration? n
* Simultaneous AP and managed mode? n
* Onboard wireless chipset or external adapter?  onboard
* Other software or services running with RaspAP? n

**Steps to reproduce**
So I fixed the problem, by chmod 777 of all files directory supposed to be removed, and did the uninstallation manually, but unfortunately the uninstalling didn't work cause I can still see the hotspot raspbwebgui .

**Expected behavior**
I expected the raspap to stop working .

**Actual behavior**
But I can still see the hotspot !

**Additional context**
Now I'm asking for your help and it's quite urgent I lost plenty of time at the uninstalling and fixing to default, and it's supposed to be a school assignment I'm very late .

So, what I intend to do and I ask you if it possible :
* Can I re-install raspap directly in the sd card , cause I have a headless raspberry and I can't access to Wifi as I stated since the raspap is installed?
* How can I remove raspap once at all? and How can I be sure that It's well removed .
* Can I reset the networking settings to default from sd card, so I can reconnect to my wifi ?
* Another solution? just to avoid resetting my rpi because I started thinking seriously
at it.

Thank you,

billz, 2020-11-26:
The services managed by RaspAP (hostapd, dnsmasq, dhcpcd, etc.) are still running after you execute the uninstaller. Rebooting is the easiest solution.

closed 2020-11-26 by billz

billz, 2020-11-26:
https://github.com/billz/raspap-webgui/wiki/FAQs