RaspAP/raspap-webgui #721: 090_adblock.conf stops DNS Masquerade service from starting.

edalee, 2020-11-29:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
`090_adblock.conf` stops DNS Masquerade from starting.
Having selected no to the Ad blocker on initial install and again when upgrading to latest version. The file `090_adblock.conf` is installed in the `/etc/dnsmasq.d/`. This conf expects to files to be present on the file system: 
```
conf-file=/etc/raspap/adblock/domains.txt
addn-hosts=/etc/raspap/adblock/hostnames.txt
```
However, they where never create when installing RaspAP.
This in the latest version (656e5d8), subsequently stops the DNS Masquerade service from running.

**Your environment**
* Raspberry Pi hardware Pi 4 Model B
* Raspbian GNU/Linux 10 (buster)
* Y
* Y
* Quick Install
* Y
* N
* Onboard wireless chipset
* NordVPN client 

**Steps to reproduce**
Select **no** to Add Adblocker in install step _(most likely version: 0801110)_ perhaps twice. The second **no** when upgrading to version: 656e5d8

**Expected behavior**
All services required to run when starting RaspAP should start without errors no matter the set-up option. 

**Actual behavior**
> Nov 29 22:01:11 wifi systemd[1]: Starting dnsmasq - A lightweight DHCP and caching DNS server...
> Nov 29 22:01:11 wifi dnsmasq[1008]: dnsmasq: syntax check OK.
> Nov 29 22:01:11 wifi dnsmasq[1009]: dnsmasq: cannot read /etc/raspap/adblock/domains.txt: No such file or directory
> Nov 29 22:01:11 wifi dnsmasq[1009]: cannot read /etc/raspap/adblock/domains.txt: No such file or directory
> Nov 29 22:01:11 wifi dnsmasq[1009]: FAILED to start up
> Nov 29 22:01:11 wifi systemd[1]: dnsmasq.service: Control process exited, code=exited, status=3/NOTIMPLEMENTED
> Nov 29 22:01:11 wifi systemd[1]: dnsmasq.service: Failed with result 'exit-code'.
> Nov 29 22:01:11 wifi systemd[1]: Failed to start dnsmasq - A lightweight DHCP and caching DNS server.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Easily fix by: `sudo rm /etc/dnsmasq.d/090_adblock.conf`
Or by commenting out the two lines in: `sudo vi /etc/dnsmasq.d/090_adblock.conf` thus:
> \# conf-file=/etc/raspap/adblock/domains.txt
> \# addn-hosts=/etc/raspap/adblock/hostnames.txt

billz, 2020-11-30:
Can you confirm that you a) ran the installer as normal (no -y, --yes or --assume-yes option), and b) used the `-u` switch to upgrade an existing install, again without the -y, --yes option? The file `090_adblock.conf` is only added if accepted at the install prompt, or with the unattended install option.

edalee, 2020-12-01:
First install in August was with this command: `curl -sL https://install.raspap.com | bash`. On the step/question Add adblocker (Beta) Y/N, `n` was submitted.

When upgrading – last week 30 Nov 2020 – this command: `curl -sL https://install.raspap.com | bash -s -- -u`. On the step/question Add adblocker (Beta) Y/N `n` was submitted.

billz, 2020-12-01:
Did you capture the install log? The installer will report each step, including optional components that are skipped.

edalee, 2020-12-01:
From the terminals stdout? Unfortunately not. I take it these are not by default persisted somewhere?

billz, 2020-12-02:
It is now :) If you're able to repro this with log output that would be ideal. Otherwise I'll try to look into it.

closed 2024-01-13 by github-actions[bot]