RaspAP/raspap-webgui #722: iptables rules do not survive a reboot.

mrespin, 2020-11-30:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
After fresh install without a reboot all working fine. After rebooting, HotSpot clients do not have the internet.

**Your environment**
* Rock Pi S 1GB NAND Wifi/BT 
* Raspbianr: ockpis_debian_buster_minimal_arm64_20200809_0853-gpt.img
* Raspbian version Buster Lite
* Followed the project prerequisites? (Y)
* Checked the project FAQ? (Y)
* RaspAP Quick Install?
* Using default configuration? (Y)
* Onboard wireless chipset. 
* Other software or services running with RaspAP - No

**Steps to reproduce**
Install default all working fine, after reboot iptables missing rules, clients without internet.

**Actual behaviour**
iptables missing rules show by install script:
Adding rule: -t nat -A POSTROUTING -j MASQUERADE
Adding rule: -t nat -A POSTROUTING -s 192.168.50.0/24 ! -d 192.168.50.0/24 -j MASQUERADE

**Additional context**
if I add iptables rule all working fine (till next boot):
iptables -t nat -A POSTROUTING -j MASQUERADE

when using iptables-save I got the message:
Warning: iptables-legacy tables present, use iptables-legacy-save
However, using iptabes-legacy-save creating an empty file.

billz, 2020-11-30:
Which install method did you use? If manual, did you install `iptables-persistent` first? The quick installer will raise an error if iptables-save can't be executed. 

https://github.com/billz/raspap-webgui/blob/30ed6b90269c137812c0c791a56f18db43afac51/installers/common.sh#L434

mrespin, 2020-11-30:
I installed using the recommended method: curl -sL https://install.raspap.com | bash
However, I checked that iptables-persisten is installed but:
root@rockpi-s:~# iptables-save > /etc/iptables/ipv2.v4
root@rockpi-s:~# iptables-restore < /etc/iptables/ipv2.v4
iptables-restore v1.8.2 (nf_tables):
line 3: CHAIN_UPDATE failed (No such file or directory): chain PREROUTING
line 4: CHAIN_UPDATE failed (No such file or directory): chain INPUT
line 5: CHAIN_UPDATE failed (No such file or directory): chain POSTROUTING
line 6: CHAIN_UPDATE failed (No such file or directory): chain OUTPUT
root@rockpi-s:~#

billz, 2020-11-30:
Debian support is [currently in beta](https://github.com/billz/raspap-webgui#supported-operating-systems), meaning it passed initial acceptance testing but no additional support is available. If you can share a solution for your distro I might be able to work it into the project. PRs are welcome, thanks

mrespin, 2020-11-30:
This is Armbian I pick it up as raspap says Armbian support is official not beta.

billz, 2020-11-30:
The reported image name is ambiguous. Please be specific with your OS.

mrespin, 2020-11-30:
> The reported image name is ambiguous. Please be specific with your OS.
Yes, my fault I paste wrong image name OS, already altered:
* OS: Armbian_20.11_Rockpi-s_buster_legacy_4.4.243_minimal
* image URL: https://www.armbian.com/radxa-rockpi-s/
* Armbian version Buster Lite

billz, 2020-11-30:
You are better off posting in the Armbian forums for this distro. Armbian Buster mainline kernel 5.8 was used on an OrangePi for testing. I don't have a RockPi on hand so testing on this distro won't happen anytime soon. Best of luck.

mrespin, 2020-11-30:
Armbian, which RaspAP support for is official, uses Debian 10 Buster, which dropped iptables support in favour of nftables. RaspAP still uses old iptables which Buster cannot restore them correctly (particularly saving and restoring problem).
Therefore, are you sure I shall report a problem with old iptables usage on Armbian forum?
Am I right?

billz, 2020-11-30:
Acceptance testing was done on Armbian Buster mainline kernel, not minimal. You can wait several weeks for me to get around to testing on this distro, if I don't forget about it or if something more important doesn't come up. Or you can open a PR and improve this project. Up to you.

Reminder that "supported" [is not a guarantee](https://github.com/billz/raspap-webgui#supported-operating-systems).

mrespin, 2020-12-01:
I am very happy to help to fix the problem. However, I am not sure that is the case with iptables/nftables. Do I on the right track? I did the same deployment in September using same sources (possible bit older versions) and all was working fine. Any suggestions to investigate the problem?

mrespin, 2020-12-01:
I found the solution:
Just run this command, forcing Buster to use old iptables:
update-alternatives --set iptables /usr/sbin/iptables-legacy
After that change all working fine. Thank you for your help.

closed 2020-12-01 by billz