RaspAP/raspap-webgui #723: GUI - lighttpd do not survive a reboot.

mrespin, 2020-11-30:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
After fresh install without a reboot all working fine. After rebooting, GUI is unavailable (service Lighttpd fail to start).

**Your environment**
* Rock Pi S 1GB NAND Wifi/BT 
* Raspbianr: ockpis_debian_buster_minimal_arm64_20200809_0853-gpt.img
* Raspbian version Buster Lite
* Followed the project prerequisites? (Y)
* Checked the project FAQ? (Y)
* RaspAP Quick Install?
* Using default configuration? (Y)
* Onboard wireless chipset. 
* Other software or services running with RaspAP - No

**Steps to reproduce**
Install default all working fine, after reboot GUI is not available.

**Actual behaviour**
lighttpd complain about permission problem with /var/log/lighttpd/error.log and stops after 5 attemps

**Additional context**
To fix issue pernamenty:
chown www-data:www-data /var/log/lighttpd/error.log

billz, 2020-11-30:
As with your previous issue, please indicate the install method you are using.

mrespin, 2020-11-30:
I installed using the recommended method: curl -sL https://install.raspap.com | bash

billz, 2020-11-30:
This project does not touch lighttpd's logs. You may be better off reporting this to lighttpd or checking for [similar issues](https://redmine.lighttpd.net/issues/2847) before reporting them here.

Related to #722

mrespin, 2020-11-30:
Thank you for finding the real source of the bug. 
I found the fix by "try and test" method. 
Afterwards, I did not search, just wanted to share with the community with the solution.

closed 2021-11-27 by billz