RaspAP/raspap-webgui #725: Unable to connect to the hostpot after setting bridged mode ?

jeremyfix, 2020-12-02:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**

After installing raspap with the quick installation script, logging on the webgui and enabling the bridged AP mode, I was unable to connect to the wifi hostpot with my laptop. I tried to connect to the hotspot only after having switched to AP mode. 

I'm not sure if this is a bug or if I misconfigured something in the hotspot but I did some change that ultimately allowed me to connect to it. 

I followed the steps in [1] and if I remember correctly, I did change two things : 
-  systemctl status systemd-networkd   returned me it was not running, so I triggered a sudo systemctl enable systemd-networkd
- I changed the security, in /etc/hostapd/hostapd.conf to : 
```
wpa_pairwise=TKIP
rsn_pairwise=CCMP
```
previously, i had only CCMP mentioned

Otherwise, thanks for this great , fantastic, (did I say amazing ?) work !

[1] https://www.raspberrypi.org/documentation/configuration/wireless/access-point-bridged.md

**Your environment**
* Raspberry Pi hardware  :  Pi 3 Model B+
* Raspbian version : 
* Followed the project prerequisites?  Y
* Checked the project FAQ?  Y
* RaspAP Quick Install or Manual setup?  Quick install
* Using default configuration? Y
* Simultaneous AP and managed mode? (Y/N)
* Onboard wireless chipset or external adapter?  Onboard
* Other software or services running with RaspAP? No

**Steps to reproduce**

- use the quick installer with a screen,keyboard,mouse plugged in the raspberry
- open the webgui, switch to bridged mode. 
- restart the raspberry

**Expected behavior**

I was expecting to be able to connect to the hotspot and get an IP from my home network on which the rapsberry is connected with eth0 (from which it got an IP)

**Actual behavior**

From my laptop (Ubuntu 18.04), I saw the wifi SSID listed but clicking on it  failed to get me connecting on it. I'm not sure about the reason but after some time, my wifi client finally connected to another wifi. 

**Screenshots**

**Additional context**

Panda-3, 2020-12-06:
I have had the same issue for quite a while, I have not yet been able to fix it and currently just use it without bridge mode

billz, 2020-12-06:
RaspAP [delegates all DHCP control to your router](https://github.com/billz/raspap-webgui/wiki/Bridged-AP-mode) in bridged AP mode.

If you have trouble connecting clients and suspect a problem with hostapd, start with this project's [default configuration](https://github.com/billz/raspap-webgui/wiki/Reporting-issues#default-settings) in routed AP mode _first_. Enabling logging for DHCP and hostapd can help you identify problems. If you have no issues with client connectivity with the default routed AP, but cannot connect clients in bridged AP mode, the problem lies with your router—not RaspAP. Check your router's web interface and DHCP settings.

Troubleshooting client connectivity in bridged AP mode is not supported here. No hard feelings.

billz, 2020-12-06:
Refer to #712 for more info

closed 2021-03-27 by billz