RaspAP/raspap-webgui #726: WIFI-QR-Code still accessable

CrashPlayer, 2020-12-06:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
Like in #622 the QR-Code is obtainable without logging in

**Your environment**
* Raspberry Pi hardware = Raspberry Pi 3B
* Raspbian version Buster Desktop or Buster Lite
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install or Manual setup? Quick Installer
* Using default configuration? Y
* Simultaneous AP and managed mode? N
* Onboard wireless chipset or external adapter? Onboard
* Other software or services running with RaspAP? N

**Steps to reproduce**
Load the startpage except you dont input, but you click cancel so it shows you "Not authorized". Then you edit the HTML of the page's body using the inspect function (tested in Firefox or Chromium/Chrome) and you replace body's inner HTML with this:
`<img src="/app/img/wifi-qr-code.php" class="figure-img img-fluid" alt="RaspAP Wifi QR code" style="width:100%;">`
That's the original HTML code used in templates/hostapd/security.php to call the QR-Code.

**Expected behavior**
QR-Code not loaded

**Actual behavior**
QR-Code visible

**Screenshots**
Not necessary.

**Additional context**
Weather you log in or not, the QR-Code PHP doesn't check the credentials, but checks if it's being called directly or through another document. So you could call it from within any document, even from another server. Maybe you could use the QR-Code-PHP-File to generate not the image but the Data-URL for the QR-Code (Base64) and make it only usable from the PHP-Skripts on the server itself.

billz, 2020-12-06:
Duplicate of #622

closed 2020-12-06 by billz

CrashPlayer, 2020-12-07:
No, #622 is closed, but in other ways still there and therefore a severe vulnerability

billz, 2020-12-08:
Better to reopen the existing issue than create a new one, thx