RaspAP/raspap-webgui #733: No client wifi config possible after first hotspot start, no routing (Pi Zero)

rxxty, 2020-12-15:
Hello,

i know there are many Pi Zero related issues, but i couldn find mine yet, as i think i followed the instructions precisely.

**Describe the bug**
Client wifi networks are no longer visible after Hotspot configuration/activation
No traffic is going trough from AP to client at any time.

**Your environment**
* Raspberry Pi hardware: Pi Zero W)  
* Raspbian version examples: RPi OS Lite, Stretch+Buster
* Followed the project prerequisites? Y)
* Checked the project FAQ? Y)
* RaspAP Quick Install or Manual setup? Quick
* Using default configuration? Y
* Simultaneous AP and managed mode? Y
* Onboard wireless chipset or external adapter? Onboard
* Other software or services running with RaspAP? dwc2 (ssh over usb)

**Steps to reproduce**
1. New install of newest Stretch or Buster release (+ ssh, wpa_supplicant.conf), upgrade, quick automated install, webinterface, no reboot.
2. Now I'm free to scan and select my wifi networks, top right hand corner info says "wlan0". Everything fine. Changing networks works, internet connection works.
3. Switching Wifi client AP mode to "on", Interface "wlan0". Save, Start hotspot.
4. Hotspot menu: From now on, not matter what is selected as hotspot "interface" and saving, when reloading menu changed back to "uap0".
5. Client menu: Right hand corner says "uap0". No wifi networks visible except for the one still active. If i disconnect, this "last" 
network vanishes as well.
6. After reboot, this behaviour is still the same.

**Expected behavior**
Client menu show wifi networks and lets connect to them.
Traffic is passed through between AP / client.

**Actual behavior**
See "reproduce": Everyt selection seems to snap back to "uap0" instead of "wlan0".
Client wifi networks not visible.
No traffic is going trough from AP to client.
Though: AP is on and connectable, wlan0 is sucessfully connected to the internet.

**Screenshots**
**Additional context**

I'm nowhere near a linux pro, so please say how i can gather more useful information.

Regards!

billz, 2020-12-15:
https://github.com/billz/raspap-webgui/wiki/RPi-Zero-W-AP-STA-mode

billz, 2020-12-15:
Re-read this more carefully and I see this occurs when switching out of AP-STA. Possibly fixed by #734

billz, 2020-12-16:
Resolved by #734

closed 2020-12-16 by billz