RaspAP/raspap-webgui #747: import of ovpn File is not possible

Erik381, 2021-01-03:
**Before submitting an issue**

All submitters MUST read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

Have a question or want to suggest a feature? You may do so here https://github.com/billz/raspap-webgui/discussions

* [+ ] This is a bug report
* [+] I searched existing issues before opening this one
* [+] I checked the FAQ before creating this issue
* [+] I have read and understand the issue reporting guidelines 

**Describe the bug**
A clear and concise description of what the bug is.

**Your environment**
* Raspberry Pi hardware (examples: Pi 3 Model B+)  
* Raspbian version (examples: RPi OS Lite)
* Followed the project prerequisites? (Y)
* Checked the project FAQ? (Y)
* RaspAP Quick Install or Manual setup? (Quick Install)
* Using default configuration? (Y)
* Simultaneous AP and managed mode? (Y/N)
* Onboard wireless chipset or external adapter? (Onboard)
* Other software or services running with RaspAP? (No except the background processes of latest raspian buster lite)

**Steps to reproduce**
connect to raspberryPi via SSH from Ubuntu 20.10
I started with the installation with this command: wget -q https://git.io/voEUQ -O /tmp/raspap && bash /tmp/raspap
Secondly I ran threw the installer and accepted everything with yes
After installation was completed I rebooted the raspberryPi
After the installation I connected to the WebGui using the IP of the raspberryPi because my Ubuntu runs on different network
Then I renamed the SSID and changed password for login
Now I headed to the VPN section, typed in my credentials and tried to upload my ovpn file from the Ubuntu file system. 
Nothing changed.
I tried to doubble click in file explorer on the file and to use the open button in file explorer. Secondly I tried to turn off Hotspot and Vpn but the bug still exists. 
I tried to connect to the WebGui using Ubuntu 20.10, Windows 10, IOS 14.3 and tried to upload the file from these OS 's. However no success there either.

**Expected behavior**
Path to file / file name should be set into the ovpn file selection field. The file should upload to the WebGui 

**Actual behavior**
File name / path is not set and no file is uploaded. 

**Screenshots**
Video is attached 
Open Video with VLC MediaPlayer when using Linux Ubuntu

https://user-images.githubusercontent.com/43499018/103487964-5f91e080-4e09-11eb-9c69-be80f827022a.mp4

**Additional context**
Add any other context about the problem here.

billz, 2021-01-03:
Ubuntu 20.10 is unsupported. See the project readme.
PRs are welcome if you'd like to improve this project for your OS.

Erik381, 2021-01-04:
Ubuntu could be a problem though.  However I used the webGUI with Windows, Ubuntu, IOS and it's still the same bug. If you want I can help improve this bug and to improve Ubuntu 20.10 support.

billz, 2021-01-04:
Your video displays "OpenVPN client.conf uploaded successfully" so your file imported. Several error cases are handled in the event an .ovpn file fails to upload:

https://github.com/billz/raspap-webgui/blob/d500ee28998507430cdd0c88b3066b91f3eea793/includes/openvpn.php#L90-L100

None of these errors are thrown in your case. Confirm the upload with `sudo cat /etc/openvpn/client/client.conf`. 
Be aware that the filename / path is not displayed in the input and you must then choose 'Start OpenVPN'.

Erik381, 2021-01-04:
Oh now I understand. I checked everything again and followed your steps, now it works again.
Thanks for the advice

closed 2021-01-04 by billz