RaspAP/raspap-webgui #752: Wifi network and uap0 interface disappeared after changing the SSID and password of hotspot

nithin-bose, 2021-01-09:
* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
 After confirming that the client connection was working fine by connecting to the default wifi network using the default wifi password, I tried changing the SSID and password of the hot spot. Upon reboot, the wifi network disappeared. The uap0 interface was also gone. 

**Your environment**
* Raspberry Pi hardware: Pi 4, 4GB  
* Raspbian version: RPi OS Lite
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install
* Using default configuration? Y
* Simultaneous AP and managed mode? Y
* Onboard wireless chipset 
* Other software or services running with RaspAP? None, it was a fresh install

**Steps to reproduce**
1. Install RaspAP using the quick installer
2. Setup client AP mode and reboot
3. Verify that the client connection is working. (uap0 was created and working)
4. Change the SSID and/or password (I changed both). If its relevant, I had uap0 selected in the interface drop down when I did this. 
5. The Pi should still be accessible through the client network, even though the hotspot had disappeared

**Expected behavior**
The SSID/password of the hot spot should have changed and devices should have been able to connect to the AP as usual

**Actual behavior**
The Wifi network created by Rasp AP disappeared . I was able to ssh into the pi using the client wifi network and found that the uap0 interface had also disappeared.

**Additional context**
```
$ sudo systemctl status raspapd.service
● raspapd.service - RaspAP Service Daemon
   Loaded: loaded (/lib/systemd/system/raspapd.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Sat 2021-01-09 19:35:37 IST; 26s ago
  Process: 2123 ExecStart=/bin/bash /etc/raspap/hostapd/servicestart.sh --interface uap0 --seconds 3 (code=exited, status=0/SUCCESS)
 Main PID: 2123 (code=exited, status=0/SUCCESS)

Jan 09 19:35:19 raspberrypi bash[2123]: Disabling systemd-networkd
Jan 09 19:35:20 raspberrypi bash[2123]: Adding uap0 interface to uap0
Jan 09 19:35:20 raspberrypi bash[2123]: command failed: No such device (-19)
Jan 09 19:35:20 raspberrypi bash[2123]: uap0: ERROR while getting interface flags: No such device
Jan 09 19:35:20 raspberrypi bash[2123]: Starting network services...
Jan 09 19:35:20 raspberrypi bash[2123]: Job for hostapd.service failed because the control process exited with error code.
Jan 09 19:35:20 raspberrypi bash[2123]: See "systemctl status hostapd.service" and "journalctl -xe" for details.
Jan 09 19:35:37 raspberrypi bash[2123]: RaspAP service start DONE
Jan 09 19:35:37 raspberrypi systemd[1]: raspapd.service: Succeeded.
Jan 09 19:35:37 raspberrypi systemd[1]: Started RaspAP Service Daemon.
```
Trying to restart the service just kept giving the same error.

```
$ sudo systemctl status hostapd.service 
● hostapd.service - Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator
   Loaded: loaded (/lib/systemd/system/hostapd.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2021-01-09 19:34:52 IST; 8s ago
  Process: 2110 ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid -B $DAEMON_OPTS ${DAEMON_CONF} (code=exited, status=1/FAILURE)

Jan 09 19:34:52 raspberrypi systemd[1]: hostapd.service: Service RestartSec=2s expired, scheduling restart.
Jan 09 19:34:52 raspberrypi systemd[1]: hostapd.service: Scheduled restart job, restart counter is at 1.
Jan 09 19:34:52 raspberrypi systemd[1]: Stopped Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
Jan 09 19:34:52 raspberrypi systemd[1]: hostapd.service: Start request repeated too quickly.
Jan 09 19:34:52 raspberrypi systemd[1]: hostapd.service: Failed with result 'exit-code'.
Jan 09 19:34:52 raspberrypi systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
```
Trying to restart the service just kept giving the same error.

```
$ cat /etc/raspap/hostapd.ini 
WifiInterface = uap0
LogEnable = 0
WifiAPEnable = 1
BridgedEnable = 0
WifiManaged = uap0
```
Just because I felt I had nothing more to do,
```
$sudo bash servicestart.sh 
Stopping network services...
Disabling systemd-networkd
Starting network services...
Job for hostapd.service failed because the control process exited with error code.
See "systemctl status hostapd.service" and "journalctl -xe" for details.
sleep: invalid time interval ''
Try 'sleep --help' for more information.
sleep: invalid time interval ''
Try 'sleep --help' for more information.
RaspAP service start DONE
```

billz, 2021-01-10:
Changes to the SSID should be applied to the `wlan0` physical device, not `uap0` (a virtual interface). In other words, configure hostapd _before_ enabling this mode.

Read [this wiki](https://github.com/billz/raspap-webgui/wiki/RPi-Zero-W-AP-STA-mode) for an understanding of how these interfaces work in AP-STA.

nithin-bose, 2021-01-10:
Thanks for the reply.
In fact I did go through the wiki, which says:
```
In this configuration, we create a virtual network interface (here uap0) and use it instead of wlan0 on the AP side.
```
which is why I edited the uap0 settings.

Why did the interface disappear? Why did it not get created again, even when I ran hostapd.service explicitly? if I understand correctly, I guess the service uses the servicestart.sh script which seems to delete and recreate everything from scratch.

Sorry, my linux networking skills are pretty limited. Why can't hostapd be reconfigured after its enabled? Is that a UI limitation? Can I change the config files directly?

nithin-bose, 2021-01-10:
After a lot of reinstalls I got everything to work, including the VPN. But I found a DNS leak and tried to change the DNS settings, now the machines connected to the AP does not seem to get an IP. In the client mode, is it like the system is immutable?

billz, 2021-01-10:
> Why did the interface disappear? 

Jan 09 19:35:20 raspberrypi bash[2123]: Adding uap0 interface to uap0
Jan 09 19:35:20 raspberrypi bash[2123]: command failed: No such device (-19)
Jan 09 19:35:20 raspberrypi bash[2123]: uap0: ERROR while getting interface flags: No such device

Saving a new hostapd configuration while already in AP-STA mode sets `WifiInterface = uap0`. This results in an invalid config which leads to the above errors. The reason for this is RaspAP has no way to determine which physical interface you want to enable with `uap0` (wlan0 is the default, but there could be other adapters present). The wiki page on AP-STA doesn't mention this, so I've updated it. 

> Why can't hostapd be reconfigured after its enabled?

Hostapd can be reconfigured at any time (even while it's running) with the exception of the 'special' AP-STA mode. I thought of disabling some parts of the UI in this case, but think this could confuse users. The solution, for now, is to stop hostapd, disable AP-STA, make your config changes on `wlan0`, re-enable AP-STA and start hostapd.

nithin-bose, 2021-01-11:
@billz Thank you for the explanation.

> The solution, for now, is to stop hostapd, disable AP-STA, make your config changes on wlan0, re-enable AP-STA and start hostapd.

It would be great if you can add this to the FAQ.

Is there a way to revert to the previous working config, even if its manually? I have like reinstalled RaspAP at least 15 times over the weekend.  

Suggestion: It took me a few reinstall to understand the client mode had to be enabled before reboot. The note under "Simultaneous AP and Wifi client" seems to be clear now that I have got it working. But weirdly, it wasn't when I installing RaspAP. Maybe you could add a line at the end of the quick install process?

Inspite of the rough edges, RaspAP is a great software, thanks a lot for all the effort you have put in.

closed 2021-01-12 by billz