RaspAP/raspap-webgui #775: OpenVPN Config with Passphrase

githubtobi, 2021-01-20:
* [X] This is a bug report
* [X] I searched existing issues before opening this one
* [X] I checked the FAQ and Wiki before creating this issue
* [X] I have read and understand the issue reporting guidelines 

**Describe the bug**
- Uploaded OpenVPN Config with config line: "askpass /etc/openvpn/phrase.txt"
- Added passphrase.txt to /etc/openvpn/phrase.txt
- OpenVPN is not connecting but a message appears in the terminal : "Password entry required for 'Enter Private Key Password:' (PID 1950). Please enter password with the systemd-tty-ask-password-agent tool:"
- After manually rentering the PW via systemd-tty-ask-password-agent it's still not connecting

**Your environment**
* Raspberry Pi 3 Model B+
* RPi OS Lite
* RaspAP version 2.52
* RaspAP Quick Install
* Using default configuration? Y
* Onboard wireless chipset or external adapter? WLAN0 (internal) as Client, WLAN1 (USB Stick) as Hotspot (works fine)

**Steps to reproduce**
- clean install
- added openvpn config file
- added openvpn phrase.txt file (just one line: the passphrase)
- clicked the "start openvpn" button

**Expected behavior**
- openvpn to add TUN0 interface and establish the connection

**Actual behavior**
- openvpn is asking for the password via the command line. but even after entering the passphrase, openvpn is not connecting

**Additional context**
Journal Log:
Jan 20 17:40:44 raspap openvpn[1189]: Outgoing Control Channel Encryption: Cipher 'AES-256-CTR' initialized with 256 bit key
Jan 20 17:40:44 raspap openvpn[1189]: Outgoing Control Channel Encryption: Using 256 bit message hash 'SHA256' for HMAC authentication
Jan 20 17:40:44 raspap openvpn[1189]: Incoming Control Channel Encryption: Cipher 'AES-256-CTR' initialized with 256 bit key
Jan 20 17:40:44 raspap openvpn[1189]: Incoming Control Channel Encryption: Using 256 bit message hash 'SHA256' for HMAC authentication
Jan 20 17:40:44 raspap openvpn[1189]: TCP/UDP: Preserving recently used remote address: [AF_INET]XXX:YYY
Jan 20 17:40:44 raspap openvpn[1189]: Socket Buffers: R=[131072->131072] S=[16384->16384]
Jan 20 17:40:44 raspap openvpn[1189]: Attempting to establish TCP connection with [AF_INET]XXX:YYY [nonblock]

-> now the broadcast message appears 

	Password entry required for 'Enter Private Key Password:' (PID 1221).
	Please enter password with the systemd-tty-ask-password-agent tool:

After I have entered the passphrase journal shows the following:

Jan 20 17:40:48 raspap openvpn[1189]: TCP: connect to [AF_INET]XXX:YYY failed: No route to host
Jan 20 17:40:48 raspap openvpn[1189]: SIGUSR1[connection failed(soft),init_instance] received, process restarting

billz, 2021-01-20:
> OpenVPN is not connecting but a message appears in the terminal : "Password entry required for 'Enter Private Key Password:'

This is expected from your config. From the openvpn man page:

> For  the  extremely  security  conscious, it is possible to protect your private key with a password.  Of course this means that every time the OpenVPN daemon is started you must be there to type the password.  The --askpass option allows you to start OpenVPN from the command line. It will query you for a password before it daemonizes.  

Storing your password in a file invalidates the extra security of an encrypted private key, so it is not implemented in this project. If this is your own VPN server, reconfigure it to use the `--auth-user-pass` option instead of `--askpass`.

githubtobi, 2021-01-21:
thank you. solved!

closed 2021-01-21 by githubtobi