RaspAP/raspap-webgui #784: Connecting to hotspot do not contiue with "Obtaining IP address..."

McTom234, 2021-01-24:
* [ ] This is a bug report
* [X] I searched existing issues before opening this one
* [X] I checked the FAQ and Wiki before creating this issue
* [X] I have read and understand the issue-reporting guidelines 

**Describe the bug**
The bug is a connecting issue, which may lead to a bug. I have no idea if its a bug or a problem on my side. I checked the internet and issues.
I try to connect to my hotspot on my rasp pi and every device tries to connect for a couple of minutes until I stop it manually. Some devices show "Obtaining IP address".

**Your environment**
* Raspberry Pi hardware (Pi 4 Model B) 
* Raspbian version (examples: RPi Buster)
* RaspAP version - v2.5.2
* Followed the project prerequisites? Yes
* Checked the project FAQ? Yes
* RaspAP Quick Install or Manual setup? Quick Install
* Using default configuration? Y
* Simultaneous AP and managed mode? N
* Onboard wireless chipset or external adapter? Onboard wireless chipset
* Other software or services running with RaspAP? None

**Steps to reproduce**
I did not make any changes. I used the quick installer and the issue occurs. I had the problem after updating my rasp pi using v2.2. I thought about incompatible OS and RaspAP. Now I am using v2.5.2 and the issue still occurs.
After I used the installer I tried to connect to the hotspot using multiple devices. They all got the same error and I cannot connect to the hotspot. Some devices show "Obtaining IP address", others not.

**Expected behaviour**
I expect the devices connect to the hotspot.

**Actual behaviour**
The devices are trying to connect to the hotspot, do not show an error message and I have to select another hotspot to connect manually.

I know there is a similar issue report, but it is a few months ago.

billz, 2021-01-24:
Duplicate of #253

McTom234, 2021-01-24:
I checked `dnsmasq status` and `ip a`. The interface wlan0 exists and has `state UP`. So far I understand, dnsmasq started without errors:
```
dnsmasq.service - dnsmasq - A lightweight DHCP and caching DNS server
   Loaded: loaded (/lib/systemd/system/dnsmasq.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2021-01-24 13:52:04 CET; 7h ago
  Process: 8163 ExecStartPre=/usr/sbin/dnsmasq --test (code=exited, status=0/SUCCESS)
  Process: 8164 ExecStart=/etc/init.d/dnsmasq systemd-exec (code=exited, status=0/SUCCESS)
  Process: 8173 ExecStartPost=/etc/init.d/dnsmasq systemd-start-resolvconf (code=exited, status=0/SUCCESS)
 Main PID: 8172 (dnsmasq)
    Tasks: 1 (limit: 3861)
   CGroup: /system.slice/dnsmasq.service
           └─8172 /usr/sbin/dnsmasq -x /run/dnsmasq/dnsmasq.pid -u dnsmasq -r /run/dnsmasq/resolv.conf -7 /etc/dnsmasq.d,.dpkg-dist,.dpkg-old,.dpkg-new --local-service --trust-anchor=.,20326,8,2,e06d44b80b8f1d39a95c0b0d7c65d08458e880409bbc683457104237c7f8ec8d

Jan 24 13:52:03 raspberrypi dnsmasq[8172]: gestartet, Version 2.80, Cachegröße 150
Jan 24 13:52:03 raspberrypi dnsmasq[8172]: Übersetzungsoptionen: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Jan 24 13:52:03 raspberrypi dnsmasq-dhcp[8172]: DHCP, IP-Bereich 10.3.141.50 -- 10.3.141.255, Lease Zeit 12h
Jan 24 13:52:03 raspberrypi dnsmasq-dhcp[8172]: DHCP, IP-Bereich 10.3.141.50 -- 10.3.141.255, Lease Zeit 12h
Jan 24 13:52:04 raspberrypi dnsmasq[8173]: Too few arguments.
Jan 24 13:52:04 raspberrypi systemd[1]: Started dnsmasq - A lightweight DHCP and caching DNS server.
Jan 24 13:52:04 raspberrypi dnsmasq[8172]: lese /run/dnsmasq/resolv.conf
Jan 24 13:52:04 raspberrypi dnsmasq[8172]: Benutze Namensserver 192.168.178.1#53
Jan 24 13:52:04 raspberrypi dnsmasq[8172]: /etc/hosts gelesen - 8 Adressen
Jan 24 13:52:05 raspberrypi dnsmasq[8172]: /etc/raspap/adblock/hostnames.txt gelesen - 209656 Adressen
```

I have no other network applications installed, except for the webserver of course.

Thank you for referring this issue. Unfortunately, it does not solve my problem.

billz, 2021-01-25:
Discussed in detail here https://docs.raspap.com/faq/#noip

Please be aware that this basic function has been rigorously tested and validated in many thousands of installations.

McTom234, 2021-01-25:
I am sorry. I missed my ufw installation. It caused the error when connecting to the hotspot.
Maybe it would be helpful to include this hint in your documentation, although it is documented by dnsmasq.
The issue can be closed and thank you for your help. Sorry again, this is an error I made.

closed 2021-01-25 by McTom234