RaspAP/raspap-webgui #798: Bad version number?

Cotopaxi42, 2021-02-02:
Hi there,

I just updated RaspAP to the latest version with this command: curl -sL https://install.raspap.com | bash -s -- -u and it worked without problem, it seems. 
Then I rebooted my Pi 4 as asked at the end of the installer but when I go to the admin interface, it still says that the version I'm using is 2.5 instead of 2.6 on the About page. 
Did I do something wrong or is it just a wrong displayed version number?

Thanks!

billz, 2021-02-08:
I couldn't reproduce this one. From the default install location `/var/www/html`:

```
sudo git fetch --tags
sudo git checkout 2.5
```
[image]

Ran the upgrade as normal `curl -sL https://install.raspap.com | bash -s -- --upgrade` and checked the About page.

[image]

Maybe you have multiple installations?

Cotopaxi42, 2021-02-09:
I don't think so. I tried to check out as well, here is my terminal (sorry...).
```

pi@raspberrypi:/var/www/html $ sudo git fetch --tags
remote: Enumerating objects: 17334, done.
remote: Counting objects: 100% (17334/17334), done.
remote: Compressing objects: 100% (9120/9120), done.
remote: Total 17151 (delta 7056), reused 17031 (delta 6938), pack-reused 0
Receiving objects: 100% (17151/17151), 15.97 MiB | 718.00 KiB/s, done.
Resolving deltas: 100% (7056/7056), completed with 93 local objects.
From https://github.com/billz/raspap
 * [new tag]         1.3.1      -> 1.3.1
 * [new tag]         1.4.0      -> 1.4.0
 * [new tag]         1.4.1      -> 1.4.1
 * [new tag]         1.5        -> 1.5
 * [new tag]         1.5.1      -> 1.5.1
 * [new tag]         1.6        -> 1.6
 * [new tag]         1.6.1      -> 1.6.1
 * [new tag]         1.6.2      -> 1.6.2
 * [new tag]         2.0        -> 2.0
 * [new tag]         2.1        -> 2.1
 * [new tag]         2.2        -> 2.2
 * [new tag]         2.3        -> 2.3
 * [new tag]         2.3.1      -> 2.3.1
 * [new tag]         2.4        -> 2.4
 * [new tag]         2.4-beta   -> 2.4-beta
 * [new tag]         2.4.1      -> 2.4.1
 * [new tag]         2.5        -> 2.5
 * [new tag]         2.5.1      -> 2.5.1
 * [new tag]         2.5.2      -> 2.5.2
 * [new tag]         2.6-beta   -> 2.6-beta
 * [new tag]         v1.0       -> v1.0
 * [new tag]         v1.1       -> v1.1
 * [new tag]         v1.2.0     -> v1.2.0
 * [new tag]         v1.2.1     -> v1.2.1
 * [new tag]         v1.3.0     -> v1.3.0
pi@raspberrypi:/var/www/html $ sudo git checkout 2.5
Previous HEAD position was 62a5e423 Update release version 2.6
HEAD is now at b02660d5 Update release version
pi@raspberrypi:/var/www/html $ curl -sL https://install.raspap.com | bash -s -- --upgrade

 888888ba                              .d888888   888888ba
 88     8b                            d8     88   88     8b
a88aaaa8P' .d8888b. .d8888b. 88d888b. 88aaaaa88a a88aaaa8P
 88    8b. 88    88 Y8ooooo. 88    88 88     88   88
 88     88 88.  .88       88 88.  .88 88     88   88
 dP     dP  88888P8  88888P  88Y888P  88     88   dP
                             88
                             dP       version 2.6

The Quick Installer will guide you through a few easy steps

RaspAP Install: Configure upgrade
Detected OS: Raspbian GNU/Linux 10 (buster)
Using GitHub repository: raspap/raspap-webgui 2.6 branch
Configuration directory: /etc/raspap
lighttpd root: /var/www/html? [Y/n]: y
Upgrading lighttpd directory: /var/www/html
This will upgrade your existing install to version 2.6
Your configuration will NOT be changed
Complete upgrade with these values? [Y/n]: y
RaspAP Install: Updating sources
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:3 http://archive.raspberrypi.org/debian buster InRelease [32.8 kB]
Get:4 http://archive.raspberrypi.org/debian buster/main armhf Packages [351 kB]
Get:5 http://raspbian.raspberrypi.org/raspbian buster/non-free armhf Packages [104 kB]
Fetched 13.5 MB in 26s (520 kB/s)
Reading package lists...
RaspAP Install: Installing required packages
[... 54 lines cut ...]
Rule already exits: -A POSTROUTING -s 192.168.50.0/24 ! -d 192.168.50.0/24 -j MASQUERADE
Enable RaspAP control service (Recommended)? [Y/n]: RaspAP Install: Enabling RaspAP daemon
Disable with: sudo systemctl disable raspapd.service
[ ✓ ok ] 
RaspAP Install: Configure ad blocking (Beta)
Install ad blocking and enable list management? [Y/n]: n

RaspAP Install: Configure OpenVPN support
Install OpenVPN and enable client configuration? [Y/n]: n

RaspAP Install: Symlinked wpa_supplicant hooks for multiple wlan interfaces
RaspAP Install: Unmasking and enabling hostapd service
Synchronizing state of hostapd.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable hostapd
[ ✓ ok ] 
RaspAP Install: Installation completed
This project needs your help! Please consider supporting RaspAP on Open Collective or GitHub:

> https://opencollective.com/raspap
> https://github.com/sponsors/billz

The system needs to be rebooted as a final step. Reboot now? [y/N]: y
```

Did I make any mistakes ? I still see version 2.5 in the administration.

billz, 2021-02-09:
Still unable to repro.
Version information is read from includes/defaults.php. Check this value on your install.

closed 2021-11-27 by billz