RaspAP/raspap-webgui #803: Custom install directory not working

freakbsd, 2021-02-11:
**Before submitting an issue**

* [ ] This is a bug report
* [ ] I searched existing issues before opening this one
* [ ] I checked the FAQ and Wiki before creating this issue
* [ ] I have read and understand the issue reporting guidelines 

**Describe the bug**
Normal installation (curl -sL https://install.raspap.com | bash), I changed only the directory where the webgui will be installed.

**Your environment**

RASPAP_LATEST=2.6.1

root@Galactus:/var/www/html# uname -a
Linux Galactus 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
root@Galactus:/var/www/html# cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
root@Galactus:/var/www/html# cat /pro^C
root@Galactus:/var/www/html# pin
pinentry         pinentry-curses  ping             ping4            ping6            pinky            
root@Galactus:/var/www/html# pin^C
root@Galactus:/var/www/html# cat /proc/device-tree/model
Raspberry Pi 4 Model B Rev 1.1root@Galactus:/var/www/html# 
root@Galactus:/var/www/html# 
root@Galactus:/var/www/html# uname -a
Linux Galactus 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
root@Galactus:/var/www/html# cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
root@Galactus:/var/www/html# cat /proc/device-tree/model
Raspberry Pi 4 Model B Rev 1.1

**Steps to reproduce**
Normal installation (curl -sL https://install.raspap.com | bash), changed only the directory where the webgui will be installed.

**Expected behavior**
If default /var/www/html is used, then everything works as expected.

**Actual behavior**

See below complete install log:

root@Galactus:~# curl -sL https://install.raspap.com | bash -x
+ set -eo pipefail
+ _main
+ repo=raspap/raspap-webgui
+ _parse_params
+ assume_yes=0
+ upgrade=0
+ ovpn_option=1
+ adblock_option=1
+ :
[... 685 lines cut ...]
$HTTP["url"] =~ "^/(?!(dist|app|ajax|config)).*" ==> $HTTP["url"] =~ "^$webroot_dir/(?!(dist|app|ajax|config)).*"

After this change the interface it is accesible, but only the main page. Any item in menu gives 404 error. In lighttpd log (after enabling accesslog) I can see the following:

2021-02-11 21:01:56: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  parse_ini_file(/etc/raspap/hostapd.ini): failed to open stream: No such file or directory in /var/www/html/thanos/includes/wifi_functions.php on line 145
2021-02-11 21:01:56: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  parse_ini_file(/etc/raspap/hostapd.ini): failed to open stream: No such file or directory in /var/www/html/thanos/includes/dashboard.php on line 182

I remember that in previous versions there was an hostapd.ini, not .conf.

Please accept my sincere apologies in advance if I should open another bug report for the following issue, but i will describe it here because this one is easy.

Another problem observed for a long time (at least 1 year) (i didn't followed the versions adn/or diff them) is the fact that **session.cookie_httponly** it is not changed by installer, see below:

root@Galactus:/var/www/html# sed -i -E 's/^session\.cookie_httponly\s*=\s*(0|([O|o]ff)|([F|f]alse)|([N|n]o))\s*$/session.cookie_httponly = 1/' /etc/php/7.3/cgi/php.ini
root@Galactus:/var/www/html# grep session.cookie_httponly /etc/php/7.3/cgi/php.ini 
session.cookie_httponly =
root@Galactus:/var/www/html# 

I am not a regex expert, but my solution for this one is:

sed -i 's/session.cookie_httponly =/session.cookie_httponly = 1/g' /etc/php/7.3/cgi/php.ini

Simple.

Thank you.

billz, 2021-02-12:
URL rewrites were introduced in the latest release. @mp03 this should affect your implementation as well.

mp035, 2021-02-15:
I have just tested an install with custom install directory (/var/www/html/raspap) and it seems to work fine.   It should be noted that the user needs to make sure their lighttpd config (at /etc/lighttpd/lighttpd.conf) has the correct server.document-root setting which matches the install folder.  The installer does not change the document-root, and I don't think that it should because the most obvious use case for a custom install directory is for a shared http server, in which case it is wise to leave the lighttpd config untouched.  @billz could you weigh in on whether I am understanding this correctly?

billz, 2021-02-15:
@mp035 I believe the issue here is that prior to introducing URL rewrites, any subdir of the default /var/www/html webroot was accessible from the browser, eg., http://raspberrypi.local/raspap/ without touching the lighttpd config. @freakbsd can you confirm this?

freakbsd, 2021-02-15:
@billz @mp035 

Yes, I confirm, I never touch lighttpd.conf after installing the raspap, everything simply worked.
I just instaled 2.5.2 and as usual, i don't do anything else to lighttpd/raspap configs.

Thanks

freakbsd, 2021-02-15:
@mp035 

When installing raspap, lighttpd is also installed with default settings which means that server.document-root it is always /var/www/html.
From my initial report, see below:

-------------------------------------------------------
**I think the problem is the fact that 50-raspap-router.conf is not updated with custom install directory, something like:

$HTTP["url"] =~ "^/(?!(dist|app|ajax|config))." ==> $HTTP["url"] =~ "^$webroot_dir/(?!(dist|app|ajax|config))."

After this change the interface it is accesible, but only the main page. Any item in menu gives 404 error. In lighttpd log (after enabling accesslog) I can see the following:**

2021-02-11 21:01:56: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning: parse_ini_file(/etc/raspap/hostapd.ini): failed to open stream: No such file or directory in /var/www/html/thanos/includes/wifi_functions.php on line 145
2021-02-11 21:01:56: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning: parse_ini_file(/etc/raspap/hostapd.ini): failed to open stream: No such file or directory in /var/www/html/thanos/includes/dashboard.php on line 182
--------------------------------------------------------

As I said, i am not rewrite/regexp expert, but something it is not right here.

Thanks,

billz, 2021-02-15:
Modifying 50-raspap-router.conf as @freakbsd suggests loads the main page, but the url-rewrite fails to load other hrefs. 
The PHP warning raised by parse_ini_file is a relatively minor issue that is unrelated to this. I'll address it separately.

mp035, 2021-02-16:
Fixed in https://github.com/RaspAP/raspap-webgui/pull/749/commits/0cfdfd367529db93bc83fb2d461434f3a2bf28ac.

freakbsd, 2021-02-16:
I will test today and I will come back with updates.
Thank you @mp035

freakbsd, 2021-02-21:
Hi guys,

Apologies for late answer.

I checked https://github.com/RaspAP/raspap-webgui/tree/0cfdfd367529db93bc83fb2d461434f3a2bf28ac and I took only https://github.com/RaspAP/raspap-webgui/blob/0cfdfd367529db93bc83fb2d461434f3a2bf28ac/config/50-raspap-router.conf. (i didn't saw any other change except for installers/common.sh)

The new lighttpd configuration works, so the custom directory is fine, but no other menu links work, all links only lead to the dasboard. So in the end it's a rewrite issue? @mp035 can you please test more ?

Thank you.

billz, 2021-02-22:
Also tested and found the following https://github.com/RaspAP/raspap-webgui/pull/749#discussion_r580139320

mp035, 2021-03-02:
You need both the 50-raspap-router.conf and the changes to the installer.  The installer will modify 50-raspap-router.conf as required during the install process.

freakbsd, 2021-03-02:
@mp035 see my comment.
 I took 50-raspap-router.conf directly after checking your changes in common.sh.and I confirmed that your changes were okay, but new behavior was observed.

Ok, I wasn't clear, so I took 50-raspap-router.conf directly and, of course, adapted to my needs (REPLACE_ME => custom), so that's why I said that the new lighttpd configuration is ok  (no need to take custom.sh)

I don't think that was enough, and I see that Billz has discovered other related issues. 

Anyway, at the moment I am happy with 2.5.2 until you completely fix 2.6.x. 

Thank you for your great work @billz @mp035 .

billz, 2021-03-03:
@freakbsd thank you for reporting this and helping us to resolve it!

closed 2021-03-03 by billz

etalon, 2021-05-09:
> Hey,
> I am completely new to RaspAP and I have the same error as described. I installed RaspAP using `curl -sL https://install.raspap.com | bash` and choosed `/var/www/html/raspap` as path. Normally I use nginx as webserver, but error 404 occurs at every page but the start page. Then I tried to use lighttpd at port 81 and it is still not working.
> 
> It seems you resolved the problem, but I didn't understand. What exactly have I to do?

I am facing exactly the same error. I also have chosen the same path.

etalon, 2021-05-09:
I solved it with the comments here: https://github.com/RaspAP/raspap-webgui/discussions/866

My 50-raspap-router.conf looked like this:

`server.modules += (
	"mod_rewrite",
)

$HTTP["url"] =~ "^_raspap/(?!(dist|app|ajax|config)).*" {
    url.rewrite-once = ( "^_raspap/(.*?)(\?.+)?$"=>"_raspap/index.php/$1$2" )
    server.error-handler-404 = "_raspap/index.php"
}`

And i changed it to this:

`server.modules += (
	"mod_rewrite",
)

$HTTP["url"] =~ "^/raspap/(?!(dist|app|ajax|config)).*" {
    url.rewrite-once = ( "^/raspap/(.*?)(\?.+)?$"=>"/raspap/index.php/$1$2" )
    server.error-handler-404 = "/raspap/index.php"
}`

As you will notice it looks like that the installer is inserting the folder not with a pipe. Instead it inserted a underscore. 
You should not edit your lighttpd.conf to a other document root. Instead call the Url http://localhost/raspap/.
This worked for me.

tefracky, 2021-05-10:
Thanks a lot! First I changed the root directory of lighttpd to `/var/www/html/raspap`, but it didn't work. The root directory must be `/var/www/html` and then http://ip-address:81/raspap is working without further changes.

billz, 2021-05-10:
@mp035 are you able to repo the above?

mp035, 2021-05-17:
@billz I am unable to reproduce this.
I started with a fresh install of raspberry pi OS lite, enabled ssh and WiFi by placing the appropriate files in the /boot partition.
Logged in via SSH:
```
pi@raspberrypi:~ $ curl -sL https://install.raspap.com | bash

 888888ba                              .d888888   888888ba
 88     8b                            d8     88   88     8b
a88aaaa8P' .d8888b. .d8888b. 88d888b. 88aaaaa88a a88aaaa8P
 88    8b. 88    88 Y8ooooo. 88    88 88     88   88
 88     88 88.  .88       88 88.  .88 88     88   88
 dP     dP  88888P8  88888P  88Y888P  88     88   dP
                             88
                             dP      version 2.6.5

The Quick Installer will guide you through a few easy steps

RaspAP Install: Configure installation
Detected OS: Raspbian GNU/Linux 10 (buster)
Using GitHub repository: RaspAP/raspap-webgui 2.6.5 branch
Configuration directory: /etc/raspap
lighttpd root: /var/www/html? [Y/n]: n
Enter alternate lighttpd directory: /var/www/html/raspap
Installing lighttpd directory: /var/www/html/raspap
Complete installation with these values? [Y/n]: 
RaspAP Install: Updating sources
```

Accessing the web gui worked after install, and:
```
pi@raspberrypi:~ $ cat /etc/lighttpd/conf-enabled/50-raspap-router.conf 
server.modules += (
	"mod_rewrite",
)

$HTTP["url"] =~ "^/raspap/(?!(dist|app|ajax|config)).*" {
    url.rewrite-once = ( "^/raspap/(.*?)(\?.+)?$"=>"/raspap/index.php/$1$2" )
    server.error-handler-404 = "/raspap/index.php"
}

pi@raspberrypi:~ $ 
```

Attached is a full transcript of the install process.
[install_transcript.txt](https://github.com/RaspAP/raspap-webgui/files/6492503/install_transcript.txt)

billz, 2021-05-17:
@mp035 thanks for checking, I was also unable to repro this.
@etalon it's unclear how your file ended up with an underscore (`_`) unless it was specified during the install.

commassoc, 2021-06-04:
Hello,
I have exactly the same problem, I receive an 404 error for all main menu items in case using an alternate lighttpd root. I want to use an alternate root because I use at the same time nginx web server and nginx use also /var/www/html as a default web site. Of course I've change the port for lighttpd (=>8080). 
My solution is to use the default installation of  raspap, install nginx before raspap et move the default nginx index (index.nginx-debian.html)  to an other location. This means also modifying the default root for nginx (sites-available default) 
Best Regard,
AR

PR #813 (2021-03-03): Routing custom install dir bugfix
Cherry picked from #749 for clarity. Resolves #803