RaspAP/raspap-webgui #812: My domoticz server not reachable anymore

acca84, 2021-02-21:
**Before submitting an issue**

All submitters MUST read this first https://docs.raspap.com/issues/

Have a question or want to suggest a feature? Start or join a discussion https://github.com/raspap/raspap-webgui/discussions

* [X ] This is a bug report
* [x ] I searched existing issues before opening this one
* [ x] I checked the FAQ and Wiki before creating this issue
* [x ] I have read and understand the issue reporting guidelines 

**Describe the bug**
I installed raspap on a raspberry pi 4 running domoticz on raspbian. The domoticz server was reachable at 192.168.1.73:8080
After installing, the domoticz server is not yet reachable.
Is that possible to make them cohabitate ?

billz, 2021-02-21:
Check the Integrations section of the FAQ. Other third party software installed on the same device is unsupported

closed 2021-02-21 by billz

acca84, 2021-02-24:
Thanks
In fact the solution is to install in another location than /var/www/html and use another port than 80