RaspAP/raspap-webgui #819: WPA2-Enterprise to WPA2 AP

MrDrache333, 2021-03-08:
* [ *] This is a bug report
* [ *] I searched existing issues before opening this one
* [ *] I checked the FAQ and Wiki before creating this issue
* [ *] I have read and understand the issue reporting guidelines 

**Describe the bug**
Ive managed to get wlan1 connected to a WPA2-ENterprise WLan (FREE-RADIUS) but if I put the HotSPot on the same Channel and set it to `AP-Mode from WiFI-Client` the uap0 Interface won't get up. If I configure the hotspot without the AP-Mode from WiFi-CLient it works just fine.

**Your environment**
Raspi3b+ with latest Release of RaspAP (As of today) and everything up to date.
Extra Hardware: Edimax EW7612UAn V2

**Steps to reproduce**
Connect with the Edimax (wlan1) to a FreeRadius WPA2-Enterprise Network and try to open a HotSpot in AP-Mode from WIFI-Client

**Expected behavior**
The HotSpot should go up and I can connect with my WPA2-Network and get internet trough the WPA2-Enterprise network

**Actual behavior**
The Hotspot stays down

**Screenshots**
[image]
[image]

**Additional context**
hostapd Log:
Configuration file: /etc/hostapd/hostapd.conf (Default)
Could not read interface uap0 flags: No such device
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=uap0 disabled_11b_rates=0
Could not read interface uap0 flags: No such device
nl80211 driver initialization failed.
uap0: interface state UNINITIALIZED->DISABLED
uap0: AP-DISABLED 
uap0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface uap0 wasn't started

billz, 2021-03-14:
You are misusing AP-STA mode. With two wireless adapters, configure one as the client (as you've done already) and select the other interface for the AP side. 

https://docs.raspap.com/faq/#interfaces

MrDrache333, 2021-03-14:
As described above, i did exactly what you said. The Problem is, that RaspAP is not able to create a Hotspot of a WPA2-Enterprise.

WLan0 as Client and wlan1 as AP.

billz, 2021-03-14:
This security type is unsupported. Search the issues, it's already been requested several times.

closed 2021-11-27 by billz