RaspAP/raspap-webgui #821: Unable to install Insider.

ledge39, 2021-03-09:
**Before submitting an issue**

All submitters MUST read this first https://docs.raspap.com/issues/

Have a question or want to suggest a feature? Start or join a discussion https://github.com/raspap/raspap-webgui/discussions

* [ ] This is a bug report
* [ ] I searched existing issues before opening this one
* [ ] I checked the FAQ and Wiki before creating this issue
* [ ] I have read and understand the issue reporting guidelines 

**Describe the bug**
A clear and concise description of what the bug is.

**Your environment**
* Raspberry Pi hardware - Pi 3 Model B+  
* Raspbian version - RPi OS Lite
* RaspAP version (reported by the Quick Installer) 
* Followed the project prerequisites?  y
* Checked the project FAQ?  y
* RaspAP Quick Install 
* Using default configuration?   y
* Simultaneous AP and managed mode?  y
* Onboard wireless chipset

I have run the following:-

curl -sL https://install.raspap.com | bash -s -- --upgrade --insiders

[image]

The does not appear to beable to connect to Github directory. I have sponsered the project, so I believe it should work, please help.

closed 2021-03-09 by ledge39

billz, 2021-03-09:
Hi @ledge39, you should have received an invitation to Insiders, which gives you access to the private repo. Your invitation still has a pending status, so check for a notification from GitHub.

ledge39, 2021-03-09:
Thanks. I was being impatient .