RaspAP/raspap-webgui #826: Compute Module 4

Villmer, 2021-03-13:
Hello.

We are developing a new consumer hardware product utilizing the WiFi-enabled Compute Module 4 and a custom PCB baseboard. We are hoping to utilize RaspAP to assist users administer the device via a local hotspot. 

**History**
Previously, we were able to successfully install RaspAP (v2.5) and create a local AP on the CM4 (with a late-2020 version of the Raspberry Pi OS.)

We are now using a new version of Raspberry Pi OS (Jan 2021) in addition to a new version of RaspAP (v2.6.3). Unfortunately, the hotspot is not working out-of-the-box.

**Hardware**:
Raspberry Pi Compute Module 4:
   * 2GB Ram
   * 32 GB storage
   * WiFi / Bluetooth

We are currently using the Raspberry PI CM4 IO Board for testing.
   
**Software**
Raspberry Pi OS (32 Bit)
Release date: January 11th 2021
Kernel version: 5.4

RaspAP (v2.6.3)

The only thing we set up, prior to installing RaspAP, was our **wpa_supplicant.conf** file:

```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US
network={
	ssid="FASTWEB-91F495"
	psk="wifipassword"
}
```

We tried commenting out the network info (#) but that did not help create the local AP.

**raspap > hostapd.ini**
  
```
WifiInterface = wlan0
LogEnable = 1
WifiAPEnable = 0
BridgedEnable = 0
WifiManaged = wlan0
```

**hostapd > hostapd.conf**

```
driver=nl80211
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
beacon_int=100
auth_algs=1
wpa_key_mgmt=WPA-PSK
ssid=raspi-webgui
channel=1
hw_mode=g
wpa_passphrase=ChangeMe
interface=wlan0
wpa=2
wpa_pairwise=CCMP
country_code=US
```
Tried **interface=uap0** (didnt work)

Any time we try to save changes we get **Unable to save wifi hotspot settings**
Thought perhaps it was permissions. Tried:

```
chmod -R 777 etc/hostapd
chmod -R 777 etc/raspap
chmod -R 777 etc/wpa_supplicant
chmod -R 777 var/www/html
```

Could someone please indicate what would be the best course of action to resolve this.
I would prefer not to go back to earlier versions of Raspberry Pi OS or RaspAP, if possible.

Villmer, 2021-03-13:
SOLVED

To get AP working, we did the following:
**Step 1**
Removed the network info from wpa_supplicant.conf

Step 2
Ran thefollowing terminal commands:
```
sudo systemctl unmask hostapd
sudo systemctl enable hostapd
sudo systemctl start hostapd
```

Villmer, 2021-03-13:
Still, however, unable to save any new changes!

**Unable to save wifi hotspot settings**

This could be a permissions issue to some files and/or folders, but which?

billz, 2021-03-14:
> Ran the following terminal commands:
> sudo systemctl unmask hostapd

> Unable to save wifi hotspot settings
> This could be a permissions issue to some files and/or folders, but which?

The former is handled by the installer and specified in the manual setup instructions. The latter is probably due to an outdated `raspap.sudoers`. So we can infer that either a), the installer wasn't used, or b) the manual steps were not followed correctly.

Since your issue is missing the required template, it's impossible to determine which of the above cases applies. The template provides a baseline for diagnosing issues and saves time for everyone. Please use it.

Villmer, 2021-03-14:
All apologies for the informal format of the question.

Here is where I am now:
1. Re-flashed the CM4 with an older 12.2020 version of Raspberry Pi OS.
2. Installed RaspAP 2.5.2
3. Hotspot works when network info is commented out in wpa_supplicant.conf

However, **I cannot make save any changes** in the UI panel.

Can we please have several suggestions on how to resolve this. I believe if we can resolve the inability to save changes, we’ll be in a good position to move forward.

billz, 2021-03-14:
You're asking for someone else to give up their time and help you figure out why something isn't working. That's fine, but put in some effort first to demonstrate that it is clearly an issue with this project, rather than something else. Right now I'm not in a position to determine that without investing a bunch of effort, including purchasing a CM4 + an IO board (neither of which I have) to try to replicate the issue you're raising.

Villmer, 2021-03-14:
We’ve already done a lot of tests, as indicated in our post.
If you don’t want us to post issues, then don’t have a forum for them.
We’re not asking you to “give up your time” exclusively for us or go out
and buy anything. We’re already testing this for you. (You’re welcome)

All we asked for were some suggestions on your forum.

billz, 2021-03-14:
https://docs.raspap.com/issues/#issue-policy

tl;dr: this is not a troubleshooting forum.

closed 2021-03-14 by billz