RaspAP/raspap-webgui #833: Unable to scan for WiFi Clients

BinaryShrub, 2021-03-26:
## Checklist

- [x] This is a bug report
- [ ] I have searched this repository for existing issues
- [ ] I checked the FAQ and official documentation before creating this issue
- [ ] I have read and understand the issue reporting guidelines 

## Bug description

After going through the quickstart guide on my RPi 4 with a fresh OS lite install, I am unable to scan for networks. The only network that shows up is the previous network I setup before running through the quickstart guide.

When I go to remove the default WiFi Client I get an error saying: `Wifi settings updated but cannot restart (cannot execute "wpa_cli reconfigure")`

trying to run `wpa_client reconfigure` or `wpa_cli -i wlan0 scan` both result in `Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory` vs before the quickstart guide both commands worked.

I attempted to run the command `https://www.gitmemory.com/issue/billz/raspap-webgui/368/530141687` as described here, but it just causes my ssh terminal to freeze up and the access point to go offline, restarting the pi I see the same issue.

## Your environment
1. Operating System: RPi OS 32-bit Lite
2. Hardware and version: RPi 4
3. RaspAP version: 2.6.3
4. Followed the project prerequisites? Yes
5. RaspAP Quick Install or Manual setup? Quick Install
6. Using default configuration? Yes (without OpenVPN and Ad Blocker enabled)
7. Simultaneous AP and managed mode? Trying to, but I can't until I can connect to a WiFi Client first
8. Onboard wireless chipset or external adapter? Onboard
9. Other software or services running with RaspAP? Nothing, completely fresh

## Steps to reproduce

1. Use Raspberry Pi Installer to install RPi OS 32-bit Lite
2. Add `ssh` and `wpa_suplicant.conf` files
3. Boot
4. SSH into RPi
5. Run `wpa_cli -i wlan0 scan` and see that `OK` is returned
6. Run `sudo raspi-config` and set WLan locale to `US` and locale `US UTF-8` 
7. Run `sudo reboot`
8. Run `sudo apt-get update; sudo apt-get full-upgrade; sudo reboot`
9. Discover the access point is now available, and connect to it
10. WiFi Client shows my single client (from step 2) but discovers no other access points, and clicking connect doesnt work
11. notice on the dashboard im not connected to a WiFi Client
12. SSH into the pi and run `wpa_cli -i wlan0 scan` to see I now get `Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory` now that RaspAp is installed
13. Remove my Wifi Client from RaspAp and receive an error saying that `wpa_cli reconfigure` failed to run
14. SSH into the pii and run `wpa_cli reconfigure` to see I get the same error message (as step 12) `Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory`

billz, 2021-03-26:
It's probable you have an error in your wpa_supplicant.conf. This is discussed in the official docs https://docs.raspap.com/. Other sources such as gitmemory are obsolete and should not be used. 

1. stop hostapd 
2. no networks listed in Wifi Client?
      exec `sudo wpa_supplicant -B -Dnl80211,wext -c/etc/wpa_supplicant/wpa_supplicant.conf -iwlan0`
     `Successfully initialized wpa_supplicant`
3. rescan
4. Wifi Client > Rescan produces a list of networks
5. Enter your credentials + connect

The above works as expected on an RPi 4.

BinaryShrub, 2021-03-26:
Ugh! I figured it out... I dug through past issues and it looks to boil down to this:

If you are using `wpa_suplicant.conf` and SSH'ing into the Pi from WLan0, right after running the quick start install **DO NOT RESTART** like the guide tells you to do.

Instead, navigate to `http://raspberrypi.local/hostapd_conf` and enable `WiFi client API mode`. If you restart before doing this, the option will be "greyed" out and you are basically fubared...

Hope this helps!

Might I suggest an update to the Quick Start guide stating that if you are using WLan0 to setup, that you need to do this before hand... or in the very least an FAQ. I kept getting tripped up by this FAQ because it's the same issue, but it doesnt help because like I mentioned, the pi will freeze: https://docs.raspap.com/faq/#wifi-scanning-doesnt-work-or-i-get-the-error-cannot-execute-wpa_cli-reconfigure-help

BinaryShrub, 2021-03-26:
Reopened just so you have a chance to see how I fixed it, feel free to close after!

billz, 2021-03-26:
Ah yes, if you're using ssh to connect to the Pi on wlan0 this will be an issue. 
This is mentioned in the AP-STA docs https://docs.raspap.com/ap-sta/#when-to-reboot
However, I agree it would be helpful to clarify this in the FAQ. Thanks for pointing this out.

billz, 2021-03-26:
FAQ is updated. Thanks

closed 2021-03-26 by billz