RaspAP/raspap-webgui #855: When selected 802.11a Hotspot Wireless Mode, it does not populate channel options

ondrejsevcik, 2021-04-07:
## Checklist

- [x] This is a bug report
- [x] I have searched this repository for existing issues
- [x] I checked the FAQ and official documentation before creating this issue
- [x] I have read and understand the issue reporting guidelines 

## Bug description

## Your environment
1. Operating System: RPi OS 32-bit Lite
2. Hardware and version: RPi 4
3. RaspAP version: 2.6.3
4. Followed the project prerequisites? Unsure
5. RaspAP Quick Install or Manual setup? Quick Install
6. Using default configuration? Yes
7. Simultaneous AP and managed mode? No
8. Onboard wireless chipset or external adapter? Onboard
9. Other software or services running with RaspAP? No

## Steps to reproduce
On Hotspot setting page, when I select `802.11a - 5Ghz` option, it does not populate channels option but leaves it empty.

## Screenshots
[image]

## Additional context
I'm using clean installation of Raspbian and RaspAP, no other modifications done.

billz, 2021-04-07:
I suspect this is a browser issue. The channel select is dynamically populated depending on your selected country, as described here https://docs.raspap.com/faq/#80211ac.

Have you blocked Javascript for RaspAP, or have any plugins that might prevent JS from executing?

I'm unable to reproduce this on the latest releases of Firefox, Chrome and Safari, so I'd start by trying a different browser, or check your browser's developer console for any reported errors.

ondrejsevcik, 2021-04-07:
I've tried a few browsers (Chrome, Firefox, Safari) and it seems to be an issue in all of them.

Anyway, I've followed the FAQ and setup my WiFi to use `ac` rather than `a` and for that, the "Channel" dropdown is populated as expected. 

Thank you for help, I appreciate it.

closed 2021-04-07 by ondrejsevcik